jacopotediosi/OctoPrint-Telegram #119: Problem uploading files Telegram

Loftolle, 2018-03-13:
For some reason when i upload a larger file through Telegram (around 30mb) it does not work, it just responds with 

⚠️ Something went wrong during processing of your file.😖 Sorry. More details are in octoprint.log.

Is there a file size cap or something? it works with the regular local upload through octoprint.

aloisff, 2018-07-29:
I have the same issue (and 'too far away' to look in the octoprint.log...), my maximum file size seems to be by 20MB

Rolleroo, 2019-01-17:
Loving the telegram plugin.

Having the same issue wtih 20MB file size also. I'll post up the logs when I can log into Octoprint

fabianonline, 2019-01-18:
From the Telegram bot API description, https://core.telegram.org/bots/api#getfile: "For the moment, bots can download files of up to 20MB in size."

So yes, there is a limit and it is indeed somewhere around 20MB.

closed 2019-01-18 by fabianonline

Neeeflix, 2020-09-14:
A workaround is to zip the file before uploading ;)