jacopotediosi/OctoPrint-Telegram #122: No attribute 'username', refresh chat not working

kavabunga6, 2018-03-31:
**First problem:**
Token status is: Error: 'NoneType' object has no attribute 'username'
If revoke token and paste new, then it works before the first reboot.

**Second problem**
Refresh chat not working. Eternal spinners
If chats visible, that buttons don't work for them

Logs in attach.
P.S. sorry for my english :)

[image]
[logs.txt](https://github.com/fabianonline/OctoPrint-Telegram/files/1865346/logs.txt)

derpicknicker1, 2018-04-11:
Try to delete all telegram settings from config.yaml and restart octoprint. then configure the plugin from beginning.

kavabunga6, 2018-04-11:
@derpicknicker1 I tried, doesn't help

ChiliApple, 2018-04-11:
works for me, THX

kavabunga6, 2018-04-11:
Problem founded. Need add this parameters to config.yaml (in your chat):
- allow_users
- send_notifications
- accept_commands

all new users have no them and plugin return exception