jacopotediosi/OctoPrint-Telegram #126: socks5 proxy

clihlt, 2018-04-16:
Telegram is blocked by censorship throughout Russia. Accordingly, the bots also can not work. Please add in the settings of the bot the ability to use the  socks5 proxy.

NickRimmer, 2018-04-18:
I join you in the request,

kiryam, 2018-04-22:
plus one for that

Zoolu74, 2018-04-25:
I join! This function is needed as oxygen!

diversunt, 2018-04-27:
+1 Very need!

bumbalu, 2018-04-29:
+1 for this request

bumbalu, 2018-04-29:
This is what I get:
[image]

clihlt, 2018-05-09:
Пацаны, зырьте на пулл реквест 128 идем туда и лайкаем. Он уже вовсю работает. Надо только вручную файлы подменить и прописать в конфиге прокси вручную.

nizongdai, 2018-08-06:
+1 I need that too.

khode-mohsen, 2019-07-12:
i use this way , you just need install tor and run bot with

```bash
torify python3 bot.py
```

hallskelet, 2019-08-13:
> 
> 
> Пацаны, зырьте на пулл реквест 128 идем туда и лайкаем. Он уже вовсю работает. Надо только вручную файлы подменить и прописать в конфиге прокси вручную.

Там полплагина пришлось руками пофиксить! и я боюсь его сейчас обновлять. хз добавил ли автор в релиз этот функционал или нет.

rlogiacco, 2019-08-14:
If you guys believe what you have wrote in Cyrillic might be relevant please translate for us :smile:

clihlt, 2019-08-14:
Well 
1st cyr comment:
Guys, look at the pull request 128 we go there and like it. He is already working hard. It is only necessary to manually replace the files and register them in the proxy config manually.

Ans:
There I had to fix a half plugin with my hands! And I'm afraid to update it now. I don’t know if the author added this functionality to the release or not.

closed 2021-06-14 by giloser