jacopotediosi/OctoPrint-Telegram #127: [Feature Request] Send Message when certain Gcode is being send

luke321, 2018-04-25:
Hello,

I have been using the plugin for a long time and I really love it.
Currently I am printing a lot of multicolor prints and thought it would be useful if you can send custom messages when certain Gcodes are being send to the printer.

For example:
M600 is send to my Prusa and it goese into waiting position
Telegram sends a message that the printer is waiting

It should be possible since Octolapse can also use custom Gcode triggers for taking pictures.

Regards
Lukas

modem7, 2018-05-19:
Would definitely be useful!

jacopotediosi, 2025-07-10:
A specific notification (`gCode_M600`) for the M600 G-code is already implemented in the plugin.  
However, it is not yet possible for users to define a customizable list of G-codes to monitor.