jacopotediosi/OctoPrint-Telegram #129: I can not configure the token

adrianobarrosmba, 2018-05-01:
I already reinstalled the octoprint, but I can not configure the plugin.
show this message
Token status is: Testing the token, please wait...

Octoprint running normally, only the telegram plugin that I can not add the token.

adrianobarrosmba, 2018-05-04:
DEbug log 

```
2018-05-03 22:56:24,810 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.2.198
2018-05-03 22:56:27,814 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.2.198
2018-05-03 22:56:48,681 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2018-05-03 22:56:48,687 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2018-05-03 22:56:48,701 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-05-03 22:56:50,309 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-05-03 22:56:51,779 - octoprint.plugins.telegram - DEBUG - Testing token
2018-05-03 22:56:53,221 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2018-05-03 22:56:53,244 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-05-03 22:56:53,289 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.8 (Github)"
2018-05-03 22:56:53,298 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2018-05-03 22:57:41,356 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAA1aAaAA1aAa_a-a1A1aAA11AaaAaaAaaA

```

Atm0s, 2018-05-15:
I have same problem. Token in the log the same as topikstarter. In the settings, another Token is specified.:

Failed to establish a new connection: [Errno 101] Network is unreachable',)).  Waiting 2 minutes before trying again.)
2018-05-15 20:32:19,469 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAaAAAAA_a1aaaAaAaaaaa1aa11AAaAaA
2018-05-15 20:32:22,656 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAaAAAAA_a1aaaAaAaaaaa1aa11AAaAaA
2018-05-15 20:32:31,844 - octoprint.plugins.telegram - DEBUG - Updated chat - 217****
2018-05-15 20:32:46,833 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAaAAAAA_a1aaaAaAaaaaa1aa11AAaAaA
2018-05-15 20:33:04,868 - octoprint.plugins.telegram - DEBUG - Saving data: {u'messages': {u'PrinterStart': {u'image': True}, u'PrinterShutdown': {u'image': True}}}
2018-05-15 20:33:44,671 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAaAAAAA_a1aaaAaAaaaaa1aa11AAaAaA
2018-05-15 20:34:27,931 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAAaAAAAA_a1aaaAaAaaaaa1aa11AAaAaA
2018-05-15 20:35:21,769 - octoprint.plugins.telegram.listener - ERROR - Setting status: Got an exception while initially trying to connect to telegram (Listener not running: HTTPSConnectionPool(host='api.telegram.org', port=443): Max retries exceeded with url:

rlogiacco, 2019-07-18:
Is this still an issue?

jacopotediosi, 2025-06-24:
This issue is 7 years old. I’m unable to reproduce it with the current codebase.  
If it’s still a problem with the recently released [version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0), please open a new issue and attach the log files which you can find in *OctoPrint Settings → Logging → plugin_telegram.log*.

closed 2025-06-24 by jacopotediosi