jacopotediosi/OctoPrint-Telegram #131: Octoprint not responding to telegram inputs

GPoorten, 2018-05-06:
Setup:
Octoprint version 1.3.8
Telegram Notifications (1.4.2)
Raspberry Pi2b on Wifi

Issue
API key tests successfull. Telegram plugin does send status updates to connected user but does not respond to command inputs (/status, /files, etc....). When user is deleted, bot does not respond with "now i know you" and user does not show on user list after refresh. After reboot, user is available again. 

Observations
On the telegram setup page the following text is shown above the API key:
_"The current connection status is: Telegram API responded with code 409. Waiting 2 minutes before trying again."_

last log file
2018-05-06 22:27:00,703 - octoprint.plugins.telegram - DEBUG - Starting listener.
2018-05-06 22:27:00,714 - octoprint.plugins.telegram.listener - DEBUG - Try first connect.
2018-05-06 22:27:00,729 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: 535631637
2018-05-06 22:27:01,139 - octoprint.plugins.telegram - DEBUG - NO PHOTOS 535631637. EXIT
2018-05-06 22:27:01,167 - octoprint.plugins.telegram - DEBUG - getMe returned: {u'ok': True, u'result': {u'username': u'Phx31_Bot', u'first_name': u'Phx31', u'is_bot': True, u'id': 581848044}}
2018-05-06 22:27:01,195 - octoprint.plugins.telegram - DEBUG - getMe status code: 200
2018-05-06 22:27:01,203 - octoprint.plugins.telegram.listener - DEBUG - Setting status: Connected as @Phx31_Bot.
2018-05-06 22:27:01,211 - octoprint.plugins.telegram.listener - DEBUG - Listener is running.
2018-05-06 22:27:01,219 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 0...
2018-05-06 22:27:01,519 - octoprint.plugins.telegram.listener - DEBUG - Ignoring message because first_contact is True.
2018-05-06 22:27:01,521 - octoprint.plugins.telegram - DEBUG - Got an event: PrinterStart Payload: {}
2018-05-06 22:27:01,530 - octoprint.plugins.telegram.TMSG - DEBUG - Printer Status{'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeOrigin': None}, 'state': {'text': 'Offline', 'flags': {'cancelling': False, 'paused': False, 'operational': False, 'pausing': False, 'printing': False, 'sdReady': False, 'error': False, 'ready': False, 'closedOrError': True}}, 'currentZ': None, 'job': {'estimatedPrintTime': None, 'filament': {'volume': None, 'length': None}, 'file': {'date': None, 'origin': None, 'size': None, 'name': None, 'path': None}, 'lastPrintTime': None}, 'offsets': {}}
2018-05-06 22:27:01,532 - octoprint.plugins.telegram.TMSG - DEBUG - TEMPS - {}
2018-05-06 22:27:01,535 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: ðŸš€ Hello. I'm online and ready to receive your commands.
2018-05-06 22:27:01,538 - octoprint.plugins.telegram - DEBUG - Send_msg() found event: PrinterStart
2018-05-06 22:27:01,540 - octoprint.plugins.telegram - DEBUG - Sending a message: ðŸš€ Hello. I'm online and ready to receive your commands. with_image=True chatID= 535631637
2018-05-06 22:27:01,541 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot
2018-05-06 22:27:01,601 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2018-05-06 22:27:01,603 - octoprint.plugins.telegram - DEBUG - Sending with image.. 535631637
2018-05-06 22:27:02,060 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2018-05-06 22:27:02,063 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2018-05-06 22:27:02,344 - octoprint.plugins.telegram.listener - ERROR - Setting status: Telegram API responded with code 409. Waiting 2 minutes before trying again.
2018-05-06 22:27:09,674 - octoprint.plugins.telegram - DEBUG - Saving data: {u'chats': {u'535631637': {u'commands': {u'/help': False, u'/con': True, u'/print': True, u'No': False, u'/status': True, u'/user': True, u'/upload': True, u'/abort': True, u'/dontshutup': True, u'/ctrl': True, u'/files': True, u'/settings': True, u'Yes': False, u'/togglepause': True, u'/tune': True, u'/test': False, u'/sys': True, u'/shutup': True}, u'title': u'Guido - Poorten - @Phx31', u'image': u'/plugin/telegram/img/static/default.jpg?1525645626706', u'send_notifications': True, u'private': True, u'allow_users': False, u'notifications': {u'StatusPrinting': False, u'PrintDone': True, u'ZChange': True, u'PrinterStart': True, u'PrintFailed': True, u'StatusNotPrinting': False, u'PrintStarted': True, u'PrintResumed': True, u'PrintPaused': True, u'PrinterShutdown': True}, u'newUsr': False, u'new': False, u'id': u'535631637', u'accept_commands': True}}}
2018-05-06 22:27:58,077 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAA1Aa1AAAAaa1AAAaaAAAaaAAAAAaAAaaA
2018-05-06 22:27:58,122 - octoprint.plugins.telegram - DEBUG - Saving data: {u'chats': {u'535631637': {u'commands': {u'/help': False, u'/con': True, u'/print': True, u'No': False, u'/status': True, u'/user': True, u'/upload': True, u'/abort': True, u'/dontshutup': True, u'/ctrl': True, u'/files': True, u'/settings': True, u'Yes': False, u'/togglepause': True, u'/tune': True, u'/test': False, u'/sys': True, u'/shutup': True}, u'title': u'Guido - Poorten - @Phx31', u'image': u'/plugin/telegram/img/static/default.jpg?1525645675200', u'send_notifications': True, u'private': True, u'allow_users': False, u'notifications': {u'StatusPrinting': False, u'PrintDone': True, u'ZChange': True, u'PrinterStart': True, u'PrintFailed': True, u'StatusNotPrinting': False, u'PrintStarted': True, u'PrintResumed': True, u'PrintPaused': True, u'PrinterShutdown': True}, u'newUsr': False, u'new': False, u'id': u'535631637', u'accept_commands': True}}}
2018-05-06 22:28:17,457 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAA1Aa1AAAAaa1AAAaaAAAaaAAAAAaAAaaA
2018-05-06 22:28:34,015 - octoprint.plugins.telegram - DEBUG - Updated chat - 535631637
2018-05-06 22:29:02,358 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2018-05-06 22:31:05,770 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2018-05-06 22:32:30,254 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAA1Aa1AAAAaa1AAAaaAAAaaAAAAAaAAaaA
2018-05-06 22:33:12,305 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2018-05-06 22:35:15,705 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...

jordips, 2021-03-27:
Hi @GPoorten,

I'm sure you found the solution but just in case: this happened to me and the problem was reuse a bot that was linked to another automation with commands (home assistant). Seems that a bot can't be linked to more than one listener that sets commands to the bot (I'm not sure). Just creating a new bot for Octoprint, started to work without problems.

jacopotediosi, 2025-07-10:
`Telegram API responded with code 409` in your logs suggests a misuse on your end. Please don't reuse the same bot token for multiple instances.

closed 2025-07-10 by jacopotediosi