jacopotediosi/OctoPrint-Telegram #132: turn lights on, before sending status message with camera

Stiftchen, 2018-05-23:
Hello,
i attached a light to my Printer wich can be controlled by a Relais and the raspberry Pi Zero W.
i can activate it by using the pull down menue where you also find the reboot and so on.
this is implemented in the config.yaml of octoprint.
Now i can control the lights also with telegram and the /sys command.

now i would like to have the light on, when telegram takes a photo for the Status message (now prinitng at z=10...) and after taking the photo turn of the light again.
since the Response of the light is weak a time wait step would be very good.

this would get better pictures even at night ;-)

since i am not good at coding, how can i do this?

regards
Stiftchen

xinuxnt, 2018-11-02:
First of all, Thank you Fabian so much to do so useful plugin!! I was looking for Stiftchen request (it should marked like UPGRADE not as issue of course)  on many forums, other pkugins and so I´m on the same perspective as Stiftchen   just to have a way to run /sys Command on the send setup would be great, after that the consumer can do whatever it want with it (Turn on camera lights it`s the first thing I can imagine),   I guess a lot of users of your plugin will be very happy to see that new premium function.

giloser, 2021-02-09:
see #338

closed 2021-02-09 by giloser