jacopotediosi/OctoPrint-Telegram #135: the bot answers constantly

aloisff, 2018-06-05:
I am outwards working, my printer is not printing anyway. Only for a demonstration i send an /status to my printer at home, but now the printer send constantly status-messeages and I can't stop it from here.

upD8R, 2018-06-16:
I'm just experiencing the same, also started after sending a /status command. Not sure, if this is really an issue with the plugin, maybe it's caused by the central bot control?!

As I'm currently printing I can't try much, just added "api.telegram.org" to /etc/hosts pointing to localhost.

Never had this issue before . how did you solve it (if ever)?

aloisff, 2018-06-17:
Hello upD8R, 
after an reboot the plugin works correct, so I think it is not really an problem inside the plugin.
But maybe it is possible to add an /reboot-command to the plugin.

aloisff, 2018-08-12:
#110 seem to be the same issue

fabianonline, 2019-03-26:
This is indeed a duplicate of #110. I'll close this issue in favor of the other one.

closed 2019-03-26 by fabianonline