jacopotediosi/OctoPrint-Telegram #146: Rotate webcam 90 degrees counter clockwise

alexisml, 2018-08-01:
When you check "Rotate webcam 90 degrees counter clockwise" under OctoPrint settings (Webcam & Timelapse) the image that appears in the telegram chat is flipped 90 degrees (**NOT counter**) clockwise, so it shows the image upsidedown. This should be a simple fix.

[image]

Environment info:
- Raspberry Pi 3+
- OctoPrint 1.3.9 running on OctoPi 0.15.1
- I'm using a raspberry camera (not usb)

kaltgift, 2018-08-14:
I'd like to confirm, except I'm seeing this on a Logitech C270 usb camera.

Raspberry Pi 3+, OctoPrint 1.3.8 running on 0.14.0

rlogiacco, 2019-07-18:
I've added a few log lines at debug level which are intended to track the image transformations: would you please enable debug logging of the plugin and report the values traced into the log?

It should look like:

```
2019-07-18 01:40:53,838 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot
2019-07-18 01:40:54,027 - octoprint.plugins.telegram - DEBUG - Image transformations [H:False, V:False, R:False]

```

closed 2019-08-07 by derpicknicker1

f4nu, 2020-03-28:
@derpicknicker1 the issue was reverted, why did you close it?
[image]

As this issue is still live, can we reopen it? I'm seeing my webcam set up correctly in the _Control tab_ on Octoprint, but the images and GIFs I receive on Telegram are mirrored:

### Control tab
[image]

### Telegram
[image]

@rlogiacco I have collected the logs you put in previously:
```
2020-03-28 21:35:09,157 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot
2020-03-28 21:35:09,242 - octoprint.plugins.telegram - DEBUG - Image transformations [H:False, V:False, R:True]
```

I can confirm that the horizontal mirroring is indeed flipped; I activated it on Octoprint and the message sent on Telegram is inverted, although now correct.

[image]

[image]

codyg, 2020-03-28:
Can confirm I'm have the exact same problem! Looks right in Control looks 180 degrees off in Telegram.

f4nu, 2020-03-28:
I just noticed that the vertical flipping is indeed also flipped.