jacopotediosi/OctoPrint-Telegram #149: Status Z height

Bishamon1987, 2018-08-21:
Hey there,
When using /status the Z height is always -0.1.
So there is a error there.
Can anyone fix it?
Cheers

mmk1994, 2019-03-19:
Same Problem here - but with -0.2
And I also do not get the automatic update for Z or time

Is there any solution?

Rubines, 2020-02-14:
Stil a problem but now I get Printing at Z=0.0 allways. 
Octoprint with Prusa I 3

DavyGab, 2020-05-14:
Same problem with -0.2 but I receive a lot of messages ! Every 30 seconds...

FaFyle, 2021-02-22:
Same problem here. Has anyone found a solution since then ?

Rubines, 2021-02-22:
No I figured it out myself.

You have to print via octoprint to get that functionallity.
When you start printing via the printer interface you don't get any Z updates.