jacopotediosi/OctoPrint-Telegram #156: telegram plugin is changing the config which causes interface reloads

Mirarkitty, 2018-12-18:
The telegram plugin is changing the config,

```pi@octopi:~/.octoprint$ ls -l config.yaml*
-rw-rw-r-- 1 pi pi 4579 Dec 18 06:21 config.yaml
-rw-r--r-- 1 pi pi 4579 Dec 17 21:18 config.yaml.bup
pi@octopi:~/.octoprint$ diff config.yaml config.yaml.bup 
143c143
<         image: /plugin/telegram/img/user/pic75232830.jpg?1545114108170
---
>         image: /plugin/telegram/img/user/pic75232830.jpg?1545065971005
```

This change to `config.yaml` is propagating to the interface, which then is forcing a reload of the interface because of a new config hash. See 
https://github.com/foosel/OctoPrint/issues/2950

foosel, 2018-12-18:
As a side note, the settings are really the wrong place to store volatile data (e.g. a constantly changing URL with a cache buster). I'd rather suggest to only store the base path and then apply the cache buster in JS.

Mirarkitty, 2019-02-07:
Just removing the part that adds `"?"+time` works to solve the immediate issue.

commit ec544abb (2019-07-17): fixes #156
closed 2019-07-17 by rlogiacco