jacopotediosi/OctoPrint-Telegram #157: Commands not work if camera is down

SpawW, 2018-12-29:
Hi,

   I found a strange issue... I am runing octoprint inside a docker container and ... if the webcam container is down... the status messages (start print, end, /status, any command) simply do no work.

dmp450, 2019-04-14:
I had a similar problem. I didn't try start or end, but if I tried /status it would freeze up my bot and it wouldn't respond to any commands unless I revoked the token. 

__How I produced the problem__
I set my stream URL to be blank, and I ran into this same problem. This didn't actually occur until I restarted the Octoprint server around a week later, so I didn't know what the problem was. I found this issue, set my stream URL back to the default, restarted the server and the telegram bot works fine.

rlogiacco, 2019-07-17:
This is very probably related to a non catched exception: could you please enable telegram plugin debug logs and provide a trace of the events?

To enable debug logs, from the octoprint settings panel select _Logging_ and add (if not there already) the telegram plugin categories all at `DEBUG` level:
* `octoprint.plugins.telegram`
* `octoprint.plugins.telegram.TCMD`
* `octoprint.plugins.telegram.TMSG`

After reproducing the issue please provide the `octoprint.log` fie

leoclee, 2020-10-13:
I am seeing the same/similar issue. I don't actually have a camera set up with OctoPrint. I eventually figured out disabling "send with image" made the notifications work. Before that, I tried unchecking "Enable webcam support" and blanking out the Stream URL on the "Webcam & Timelapse" settings, but that did not work. I have attached my ocotprint.log with debug logging enabled:
[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5372643/octoprint.log)

giloser, 2020-10-13:
Can you try to install this version and set the message with image back
https://github.com/giloser/OctoPrint-Telegram/releases/download/1.6.0.b3/release.zip

thanks

commit a251992e (2020-10-13): Merge pull request #304 from giloser/master

Fix #157 issue and code cleaning
closed 2020-10-13 by giloser