jacopotediosi/OctoPrint-Telegram #159: No Notifications During Print

556duckvader, 2019-01-23:
I have notifications setup for every 60min and I am not receiving them.  I can do a /status and get the status but its not automatically sending them.

indyfansd, 2019-01-25:
I'm having the exact same issue. Does anyone have a solution for this? I've tried every interval from one minute to 1 hour and I'm not getting notifications as well

leeford, 2019-01-29:
Have you clicked the speech bubble in the 'chat' section and enabled them?

indyfansd, 2019-01-29:
Yes. All options are enabled in octoprint for telegram under chat bubble and i gave full permission to send and receive. I noticed under chat there isnt a check box for time but only for z . i have also unitsalled and made a new bot 3 times and it isnt working

556duckvader, 2019-02-19:
I have done the exact same thing as indyfansd and I am still not getting notifications during the prints without sending /status.  I get notifications at the start and end of prints but not during the prints and I have my send notification setup for 30min.

MartinRohrbach, 2019-02-26:
I had the same issue ... until I actually checked the "ZChange" notification for my user. Turns out, this is the trigger for the timed messages also so it needs to be checked for this to work. Maybe that helps you guys as well ...

If that resolves the issue the wording for that option should probably be changed.

fabianonline, 2019-03-26:
Indeed.