jacopotediosi/OctoPrint-Telegram #160: no notification via SD card print

MLWORKX, 2019-03-01:
Hi,

When printing via OctoPrint I do get start + end notifications plus time and layer hight driven notifications: great!

When saving the files on the sd card I only do get start and end notifications as the printer does not report the actual height back to Octoprint (see picture!)
http://666kb.com/i/e1oue1xhtlrr84dr3.jpg

- Everything is fine when printing directly via Octo
- I do have stl files which contain a very high number of triangels
- slicer does not reduce high triangle count therefore the USB speed is to low and I need to put those files on the SD card

...is there ANY way of telling this Plugin to message me at least every n minutes?

fabianonline, 2019-03-26:
At the moment not. The time based notification code is only triggered on z changes, so it is more a "notify on the first z change at least x minutes after the last notification". But I think having some kind of background thread sending a notification every x minutes would be possible and useful.

GilDev, 2019-06-06:
This would be nice indeed, I like printing from the SD card because of the reliability.

kanocz, 2019-08-26:
@MLWORKX  I think it's possible to implement in separate plugin
I'm working now on https://github.com/kanocz/octopi_eta_override - receiving Prusa's updates about ETA and sets it... but I can combine this with query of current position and try to emit "z-change" event with possition every minute... so if z is changed enough telegram plugin will send notification

kanocz, 2019-08-26:
@MLWORKX I think this issue can be closed now

MLWORKX, 2019-08-26:
> @MLWORKX I think this issue can be closed now

Thank you very much for your efford - amazing!

I'll test it till end of week.
I feel it should be a part of Octoprint itself without a plugin therefore I'm not sure if it should be closed even if it works like a charm (?).

kanocz, 2019-08-26:
@MLWORKX you know... maybe you're right :) my solution works only for Prusa MK3(S) printers as their slicer has extended time estimaton and notfication system... just forgot that any other printers exist :(