jacopotediosi/OctoPrint-Telegram #164: Control LED strip to switch it on, take the photo and then switch it off again

fabianonline, 2019-03-30:


MarcusWolschon, 2019-04-01:
Good idea.
"M42 S255", then take the photo, then "M42 S0"

After a print is finished, the last photo sent via Telegram of usually just black because the printer has already finished the job and switched off the lights.

xinuxnt, 2019-10-10:
Sorry, newbie question, I don`t understand, where I should add "M42 S255", then take the photo, then "M42 S0"  to do that?   Thanks in advance!

KUGA2, 2021-01-16:
I am currently working on this. Why should i do this via gcode? The LEDs are usually connected to the raspi, or why should they not?

Super simple implementation (working, but far from PR ready ;) ):
[image]

KUGA2, 2021-02-06:
Hi, I created an alternative, that replaces the included mjpg-streamer and switches the LEDs when the stream is watched. This works from Octoprint webinterface as well as for this telegram plugin.

https://github.com/KUGA2/OctoStreamer
It is not perfect. Feel free to improve or post your issues there. I will happyly look at your PRs.

giloser, 2021-02-06:
hi I have connected my led to the printer directly so I don't need that but I tried to add the functionality in the version 1.6.3.
Thanks for your work I'll still check your plugin.
let me know if the new 1.6.3 version is working for you

closed 2021-06-14 by giloser