jacopotediosi/OctoPrint-Telegram #165: "I do not understand you" in a group

Limonkufu, 2019-04-06:
Is there a way to have the bot reply to only commands starting with '/'. Otherwise in a group it always says I do not understand you, which is annoying.

rlogiacco, 2019-07-17:
We can think of two solutions: a command pair `/deaf` and `/undeaf` which make the bot ignore anything not starting with a `/` or a settings option `Commands only`...

Would you be willing to contribute such enhancement?

derpicknicker1, 2019-08-07:
I have multiple bots in groups running and they only reply to / commands

rlogiacco, 2019-08-07:
> I have multiple bots in groups running and they only reply to / commands

I miss your point

Limonkufu, 2019-08-08:
> 
> 
> > I have multiple bots in groups running and they only reply to / commands
> 
> I miss your point

Maybe he is saying our bot should only say 'I don't understand you' if the message was a command (starting with /) otherwise it should ignore the message

rlogiacco, 2019-08-08:
So, it was a +1 on this request, but why mention other bots is beyond my understanding...

closed 2025-06-25 by jacopotediosi