jacopotediosi/OctoPrint-Telegram #166: Telegram plugin disappeared from octoprint

SimoneBnc, 2019-04-13:
After an OS update I lost the telegram plugin. I tried to Uninstall it and install it several times (installation is successful every time) but it is no longer showing in octoprint and of course it is no longer working. I even tried to update octoprint to the RC11 but still no luck. Is there a way to figure out what happened?

rlogiacco, 2019-07-18:
We need to have a look at your `octoprint.log`

SimoneBnc, 2019-07-18:
Do you want it after I uninstall and reinstall the plug in or just the file after the startup?

rlogiacco, 2019-07-18:
I believe the version you tried to install is the last stable ('1.4.2') so the startup should be fine, just ensure you have the [debug logging enabled](https://github.com/fabianonline/OctoPrint-Telegram/wiki#how-do-i-enable-debug-logging) before restarting.

SimoneBnc, 2019-09-15:
I tried to install it but it does not show in the plugin page and every time I search for new plugins it always shows the INSTALL button, it successfully installs the plugin but it still missing from octorpint.

SimoneBnc, 2019-09-28:
nevermind... I made a fresh install.

closed 2019-09-28 by SimoneBnc