jacopotediosi/OctoPrint-Telegram #167: Telegram Notification Issue

556duckvader, 2019-04-15:
I am having an issue where I can see in the Octoprint logs that telegram is sending notifications but I never get them on my app.  If I restart Octoprint I get the notification that the printer is online but after that no other notifications are received.  If I do a /status it works.  Below is the log entry from a print starting.  That I never received the notification.

2019-04-15 08:01:00,065 - octoprint.plugins.telegram - DEBUG - Got an event: PrintStarted Payload: {'origin': 'local', 'name': u'MK3_BMG_PETG_TagCenter-X3.gcode', 'file': u'/home/pi/.octoprint/uploads/GTS/MK3_BMG_PETG_TagCenter-X3.gcode', 'path': u'GTS/MK3_BMG_PETG_TagCenter-X3.gcode', 'size': 4607635L, 'filename': u'MK3_BMG_PETG_TagCenter-X3.gcode'}
2019-04-15 08:01:00,081 - octoprint.plugins.telegram.TMSG - DEBUG - Printer Status{'progress': {'completion': None, 'printTimeLeftOrigin': None, 'printTime': None, 'printTimeLeft': None, 'filepos': None}, 'state': {'text': 'Printing', 'flags': {'cancelling': False, 'paused': False, 'operational': True, 'pausing': False, 'printing': True, 'resuming': False, 'sdReady': False, 'error': False, 'ready': False, 'finishing': False, 'closedOrError': False}}, 'currentZ': None, 'job': {'averagePrintTime': 21236.42766404152, 'lastPrintTime': 21236.42766404152, 'user': '_api', 'file': {'origin': 'local', 'name': u'MK3_BMG_PETG_TagCenter-X3.gcode', 'date': 1555286586, 'path': u'GTS/MK3_BMG_PETG_TagCenter-X3.gcode', 'display': u'MK3 BMG PETG_TagCenter-X3.gcode', 'size': 4607635L}, 'estimatedPrintTime': 22841.501673959094, 'filament': {u'tool0': {u'volume': 171.61512725860104, u'length': 24278.573379988058}}}, 'offsets': {}}
2019-04-15 08:01:00,081 - octoprint.plugins.telegram.TMSG - DEBUG - TEMPS - {'bed': {'actual': 17.0, 'target': 0.0, 'offset': 0}, 'tool0': {'actual': 17.3, 'target': 0.0, 'offset': 0}}
2019-04-15 08:01:00,083 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: Started printing MK3_BMG_PETG_TagCenter-X3.gcode.
2019-04-15 08:01:00,088 - octoprint.plugins.telegram - DEBUG - Send_msg() found event: PrintStarted
2019-04-15 08:01:00,091 - octoprint.plugins.telegram - DEBUG - Sending a message: Started printing MK3_BMG_PETG_TagCenter-X3.gcode. with_image=True chatID= 785070499
2019-04-15 08:01:00,091 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot
2019-04-15 08:01:00,106 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-04-15 08:01:00,130 - octoprint.plugins.telegram - DEBUG - data so far: {'parse_mode': 'Markdown', 'disable_web_page_preview': True}
2019-04-15 08:01:00,131 - octoprint.plugins.telegram - DEBUG - Sending with image.. 785070499
2019-04-15 08:01:01,276 - octoprint.plugins.telegram - DEBUG - Sending finished. 400

rlogiacco, 2019-07-17:
Is it possible your Octoprint server gets disconnected from the internet?

ThomasTL, 2019-10-14:
I have the same issue.
However, the commands: `/status`, `/gif`, `/supergif` all work fine.
The only notification I get is when I reboot the server, I get the offline notif and then online notif.
Not sure what is wrong here.

jacopotediosi, 2025-06-24:
Hi, I'm the new maintainer of this plugin.

From your logs:

> Sending finished. 400

The HTTP 400 error indicates that something is wrong with the request sent to Telegram to send the notification message.

Could you please check if the issue still occurs with [version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0)?  
If it does, please attach your logs following the instructions on [this wiki page](https://github.com/jacopotediosi/OctoPrint-Telegram/wiki/Where-are-the-logs%3F).

Thanks!

closed 2025-07-02 by jacopotediosi