jacopotediosi/OctoPrint-Telegram #174: After Octoprint upgrade to 1.3.11 telegram plugin is unable to format emo:rocket and other emos

bumbalu, 2019-05-14:
Hello, everyone. 
After Octoprint upgrade to 1.3.11 telegram plugin send the following message after octoprint start: 
===
ERROR: I was not able to forman the Notification for 'PrinterStart' properly. Please open your Octoprint settings for Telegram Notifications and check message settings for 'PrinterStart'
===
I have tried to switch on/off the markup selection or html - it did not help. Then I removed the brackets on both ends of {emo:rocket} and telegram plugin successfully formats notifications. 

We obviously can happily live without those nice emoticons but getting them back would be pleasant.  
Anybody has any ideas?

xenon2008, 2019-05-15:
I also have the same Error!

See Also my Post @Octoprint Github Post here:
https://github.com/foosel/OctoPrint/issues/3150

TheLion, 2019-05-15:
Same issue also.

VanKurt, 2019-05-16:
metoo ;-)

Evg33, 2019-05-16:
#93

Piranit, 2019-05-20:
Just delete all emo from messages.
[image]

avrahams, 2019-05-21:
same issue since upgrade to 1.3.11 - worked fine before the upgrade

⚠ ERROR: I was not able to format the Notification for 'PrinterStart' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrinterStart'.

All messages are with defult format 
PrintDone: Finished printing {file}.

Brain-Gamer, 2019-05-22:
It seems to work without any kind of issue if you use 2 spaces after the last emo instead of just one.
So instead of `{emo:rocket} Hello....` you use `{emo:rocket}  Hello....`
No idea why but this workaround fixed the issue for me.

friedl, 2019-05-22:
> 
> 
> It seems to work without any kind of issue if you use 2 spaces after the last emo instead of just one.
> So instead of `{emo:rocket}[SPACE]Hello....` you use `{emo:rocket}[SPACE][SPACE]Hello....`
> No idea why but this workaround fixed the issue for me.

Just tried this and it worked.

sagatxxx, 2019-05-24:
Same here, solved with Two spaces

Budavaril, 2019-05-26:
Found that actually "no space" or "two spaces"..

careyer, 2019-07-10:
Same here... problem still persists... workaround with TWO SPACES works

commit 9788464e (2019-07-17): fixes #174
closed 2019-07-17 by rlogiacco