jacopotediosi/OctoPrint-Telegram #176: Weird user connected to plugin

ThaliaFromPrussia, 2019-05-22:
I got a weird user in my plugin. Is this the Telegram bot? can't find any information in the readme.
Thanks in advance!

[image]

Ozkovacs, 2019-06-07:
I have the same situation, with a user called "Mekan - Bilmez / ID: 727602521 | PRIVATE".
Whenever I remove it, it gets added back automatically.
I know that I re-installed Octoprint at some point, and had to re-enable the plugin.  Can that be a reference to my older setup?!?  I doubt it, but I think it's worth mentionning.

rlogiacco, 2019-07-17:
The bots are open for public access: those are people bumping into your bots either by mistake or attempting to hack. Just be sure to refrain giving those users any permission and you will be good to go.

closed 2019-07-17 by rlogiacco