jacopotediosi/OctoPrint-Telegram #179: [Request] Manuel telegram-message from GCode

DerDominik, 2019-06-03:
Hi there,

attached a feature request for the Octoprint-Telegram-Plugin.

Would be great to send a manual telegram-message with a command from the GCode (which has to be inserted manually).

Background (for example):
With a long print and filament change during the print, it would be great to get a telegram message short before the change has to be done.
Of course you've to insert that command manually, but if the layer takes a longer time, it would be a great benefit of the plugin instead of be informed after every layer.

Doki76, 2019-06-04:
I would like to enhance this request.

I would be nice if the plugin can not just listen to a custom GC Code commands but also for a custom Response from the Printer that OctoPrint might receive.

Im using the Pluging together with PRUSAMK3 MMU2 and it would be great to get messages when a filament switch failed. The printer sends a response to the gcode for filament switches when it fails.

kind regards
Doki

MarcusWolschon, 2019-06-05:
That sounds like a job for it's own "prusa filament switcher" plugin as it surely also should pause the print and display an error in the Octoprint web UI.

Doki76, 2019-06-05:
I only need a message, everything else is already working without a plugin. 
Octoprint is waiting until the switch is successfully finished on its own.

derpicknicker1, 2019-08-08:
#127 is related to this

jacopotediosi, 2025-07-10:
This is not the same as #127.
#127 (now reopened) requested a new feature to set up a watch to monitor a custom list of G-codes and send a notification when one of them is sent by OctoPrint to the printer.

This issue, instead, is about sending custom notifications from a G-code, which is already supported by this plugin via the `UserNotif` notification (triggered when the printer sends `echo:UserNotif TEXT` over serial, e.g. from a G-code like `M118 E1 UserNotif TEXT`).