jacopotediosi/OctoPrint-Telegram #180: Setup token shows up at every page refresh

tyleraharrison, 2019-06-13:
Have used this plugin forever, today I reflashed the OctoPi SD card and installed this plugin and everything works flawlessly except for when I refresh the page I always get the token setup popup. Easily fixable by just hitting next but it's definitely annoying. It always says the token status is unknown until I check it, when it says its working just fine.

Quintin9801, 2019-06-27:
Having the exact same problem using OctoPrint on Windows instead of Pi

raidolo, 2019-09-05:
Same here, when I restart Octoprint (on docker) the wizard popup appear asking to verify the Token.

gatescs, 2019-09-29:
I just set up this plugin today and I am having the same issue.  Has anyone made any progress on resolving this?

raidolo, 2019-09-29:
For me the problem was the original Octo instance still running. Somehow when you have the plugin already started elsewhere and the bot is connected to the first instance, you can’t initiate another session properly, maybe because I was using the same chat id and the first instance “listens” for incoming messages, so the second instance could not be initiated properly. (when you use this kind of bots the plugins are always in listening mode checking for incoming messages, I didn’t check the pluginsource, but I guess it’s not different)

rlogiacco, 2019-09-30:
One code == one plugin instance

A second instance using the same code automatically kicks the second out of
play

gatescs, 2019-09-30:
Thank you for the reply.  I think I follow what you are saying, I am new to this plugin, so still learning.  So, for example, if I have multiple printers, each with their own Pi, each using Telegram, should I create a new bot for each?  Right now, I have two printers/Pi's running Telegram to test it out, but they are both using the same.  Which is likely creating the problem as you noted.

Thanks again, very much appreciated.

rlogiacco, 2019-09-30:
If you want to tell printer 1 to start printing model A and tell printer 2
to abort the print you need to “chat” with two separate bots... this
requires to create 2 bots, each one identifying a printer.

1 bot == 1 printer

Makes sense now?

rlogiacco, 2019-09-30:
Glad to hear that

closed 2019-09-30 by rlogiacco