jacopotediosi/OctoPrint-Telegram #181: Please add a notification for user interaction requests

brickkk, 2019-06-19:
Please add a notification if user interaction on the printer is necessary
(Maybe use PauseForUser Event Plugin: https://plugins.octoprint.org/plugins/PauseForUserEvent/ )

mrjackson, 2019-08-08:
This would be great for people with the Prusa MMU or anyone adding manual filament changes.

rlogiacco, 2019-08-08:
Would you guys be willing to develop this and submit a pull request? You know, this is open source software and lives of contributions

ZeGuigui, 2019-09-12:
> Would you guys be willing to develop this and submit a pull request? You know, this is open source software and lives of contributions

I have some working code, now I have to learn how to create pull requests 🤣

kanocz, 2019-09-12:
@ZeGuigui just create fork, put changes, commit, then click "pull request" :)

ZeGuigui, 2019-09-12:
> @ZeGuigui just create fork, put changes, commit, then click "pull request" :)

Done! Learned how to create and apply patches at the same time

closed 2021-02-09 by giloser