jacopotediosi/OctoPrint-Telegram #184: [Feature Request] MTProxy support

r00tGER, 2019-07-02:
Please add connection via mtproto-proxy.

rlogiacco, 2019-08-13:
Please remember this is an open source project, meaning if you desire a feature being implemented your best chance is to submit the change yourself in the form of a pull request...

hallskelet, 2019-08-13:
I remember that.
I am an engineer but not a python programmer at all.
It took me almost a week to add the annual change to SOCKS5, figuring out which lines to edit, because simply replacing the file did not work.

What do you think, how much time will it take me to deal with the fact that I don’t know anything and what I don’t know how to use?

I have angry locksmiths and pocket zippers in my bank at work and no pythons at all.

rlogiacco, 2019-08-13:
Trust me, much less than you can imagine. I had 0 experience with python and in 2 days I got a few bugs fixed and helped bring out a new release.
Anyway, I’m not asking you specifically to develop this, just asking for help as we are not paid to implement features, but I understand this might be a blocker in some countries.
In other words: I wasn’t pinpointing you, but anybody’s help is welcome :smile:

hallskelet, 2019-08-13:
Please, give me your schoolbook for Piton. :)

davtur19, 2025-06-25:
The bot does not use mtproto to communicate but the bot api, at most you could consider adding a configurable proxy in the settings, but at this point I have to ask why do you need it? And why not pass all the traffic at the system level directly behind a proxy/vpn?