jacopotediosi/OctoPrint-Telegram #187: [Feature] IMAGES for all text mensagges

Linderkin, 2019-07-26:
Implement custom images for each text field, example:

Impression failed {custom image}
Impression to Finished {custom image}

Thanks in advance

giloser, 2021-02-09:
Hi we cannot add image in the text message easily 
and what should be set as the custom image?

jacopotediosi, 2025-06-24:
For each notification message from the plugin, it’s already possible to enable sending photos and videos from the webcams.

What’s the use case for adding custom photos to the messages?  
If you want, you can already do this by specifying the URLs of static images as if they were webcams in the Multicam plugin settings.

I’m closing this issue because I don’t understand the use case.  
Feel free to provide more details and I’ll consider reopening it.