jacopotediosi/OctoPrint-Telegram #192: New /gif requires ffmpeg, but suggests to install avconv

jollino, 2019-08-08:
The new /gif and /supergif features are great, but they don't work out of the box on OctoPi, at least 0.14.0 (with OctoPrint 1.3.11) due to a hard requiremement for ffmpeg.

The images are generated correctly, but not spliced:
> 2019-08-08 11:56:47,346 - octoprint.plugins.telegram.listener - INFO - Got a command: '/gif' with parameter: '' in chat ...
> 2019-08-08 11:56:47,351 - octoprint.plugins.telegram.TCMD - INFO - Will try to create a gif
> 2019-08-08 11:56:47,372 - octoprint.plugins.telegram - INFO - Caught an exception trying clean previous images : [Errno 2] No such file or directory: '/home/pi/.octoprint/data/telegram/tmpgif/gif.mp4'
> 2019-08-08 11:56:47,374 - octoprint.plugins.telegram - INFO - will try to save image in path /home/pi/.octoprint/data/telegram/tmpgif
> 2019-08-08 11:57:01,973 - octoprint.plugins.telegram - INFO - Caught an exception trying create mp4 : [Errno 2] No such file or directory
> 2019-08-08 11:57:02,055 - octoprint.plugins.telegram - INFO - Caught an exception trying create mp4 2 : [Errno 2] No such file or directory
> 2019-08-08 11:57:02,062 - octoprint.plugins.telegram - INFO - Sending a message:  Problem creating gif, please check log file, and make sure you have installed libav-tools with command : `sudo apt-get install libav-tools` with_image=False with_gif=False chatID= 22097387

I do have avconv and libav-tools is installed, but that does not provide an ffmpeg alias; this OctoPi image in particular doesn't even carry the actual ffmpeg in the apt repository.

I looked at the code, and octoprint_telegram/__init__.py at lines 1361 does this:
`subprocess.check_call(['ffmpeg', '-r', '10', '-y', '-i' ,self.get_plugin_data_folder() + '/tmpgif/Gif_Telegram_%2d.jpg', '-crf', '20', '-g' ,'15', self.get_plugin_data_folder() + '/tmpgif/gif.mp4'])`
which obviously fails, as there's only avconv.

My workaround was to simply symlink avconv to ffmpeg in /usr/bin, but a proper fix would be to check whether ffmpeg is available, and possibly fall back to using avconv before giving up.
It should be fairly trivial to implement, but I'm currently starved of time at this stage and I never worked on PRs, so I'm afraid I just can't get to tinkering about it.

I'm not sure if any newer version of OctoPi — the image itself — installs ffmpeg proper, but mine definitely doesn't. Considering that many people just upgrade OctoPrint from within itself and leave the base OctoPi image alone, this may affect quite a few users.

(As an aside, it would be nice if the bot gave some updates while it's recording the images: right now it just sits there and users may wonder if it actually acknowledged the command.)

Thanks for the great project!

derpicknicker1, 2019-08-08:
The newest OctoPi distribution seems to have ffmpeg on board. But I agree that this has to be fixed.

fabianonline, 2019-08-08:
Regarding the feedback during creation of the gif: The bot API has the function `send_chat_action` which could be used to display a nice "<your bot> is taking a video...".

mnul, 2019-08-08:
Also, you need an option to turn it off. Right now the creation of a gif makes the Bot not responding at all. 
On a Pi Zero W it takes quite a while and blocks everything else.

rlogiacco, 2019-08-08:
> Also, you need an option to turn it off. Right now the creation of a gif makes the Bot not responding at all.
> On a Pi Zero W it takes quite a while and blocks everything else.

I agree: resource intensive tasks should be optional, disabled by default and enabled explicitly by the user, rather than the other way around.

giloser, 2019-08-09:
Is it really enable by default?
I was thinking I made it that you have to go and press the button 'with gif ' to activate it for a type of message. So the idea was not to enable it by default.
Could you confirm it to me that it is enable by default?
I'll try to wipe my plugin data and start all over again to check that.

giloser, 2019-08-09:
Hi!

I'm sorry I check and you are right I put the gif by default for the Zchange. I didn't see that and it should not be like that. I'll change that!

mnul, 2019-08-09:
A regular update through a existing octoprint setup has given me the gifs as a default. 
Worst thing is, I dont know how to deactivate it...
Please change the behaviour to being optional.

giloser, 2019-08-09:
you can desactivate it in the plugin option through octoprint by clicking the Send Gif button
[image]
to 
[image]
for the z change notification.

giloser, 2019-08-29:
Hi,

I finally made some change to try to really fix this one and for all
the version is in a pull request but if someone want to try it here is should be the link to install directly on the plugin manager
https://github.com/giloser/OctoPrint-Telegram/archive/master.zip

closed 2021-02-09 by giloser