jacopotediosi/OctoPrint-Telegram #193: Issue with /files and /print

juguerin3, 2019-08-08:
Hi,
I have a problem with the following two commands: / files and / print.
When i use /files, octoprint receive and send me the loading message : Loading files... and that's all. 
No file list or other... just the loading message.

I try to find a way to solve this error with log but i have only the following log :

2019-08-08 11:22:58,211 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 618909338...
2019-08-08 11:22:58,667 - octoprint.plugins.telegram.listener - DEBUG - Updating update_offset from 618909338 to 618909339
2019-08-08 11:22:58,669 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'message': {u'from': {u'first_name': u'Guerin', u'last_name': u'Julien', u'is_bot': False, u'id': XXXXXXXX, u'language_code': u'fr'}, u'text': u'/files', u'entities': [{u'length': 6, u'type': u'bot_command', u'offset': 0}], u'chat': {u'first_name': u'Guerin', u'last_name': u'Julien', u'type': u'private', u'id': XXXXXXXX}, u'date': 1565259778, u'message_id': 604}, u'update_id': 618909338}
2019-08-08 11:22:58,670 - octoprint.plugins.telegram.listener - DEBUG - Not changing update_offset - otherwise would reduce it from 618909339 to 618909339
2019-08-08 11:22:58,674 - octoprint.plugins.telegram.listener - INFO - Got a command: '/files' with parameter: '' in chat 434094923
2019-08-08 11:22:58,979 - octoprint.plugins.telegram - INFO - Sending a message: Loading files... with_image=False with_gif=False chatID= 434094923
2019-08-08 11:22:58,981 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True}
2019-08-08 11:22:58,982 - octoprint.plugins.telegram - DEBUG - Sending without image.. 434094923
2019-08-08 11:22:59,205 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
2019-08-08 11:22:59,281 - octoprint.plugins.telegram.TCMD - DEBUG - {'0bca257b': 'local/'}
2019-08-08 11:22:59,389 - octoprint.plugins.telegram.listener - ERROR - Exception caught! 'hash'
2019-08-08 11:22:59,391 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 618909339...

Do you have a idea to solve this bug ?

PS : This issue was here from the first install with the  v1.4.2.

Current config / plugin : 
2019-08-07 20:23:30,047 - octoprint.plugin.core - INFO - 26 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.11.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Filament Sensor Reloaded (1.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Fullscreen Plugin (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Sidebar Temp Graph (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_sidebartempgraph
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Telegram Notifications (1.5.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram
|  Tempsgraph Plugin (0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tempsgraph
|  Themeify (1.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_themeify
|  Title Status (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_title_status
|  TouchUI (0.3.13) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  TP-Link Smartplug (0.9.16) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2019-08-07 20:23:30,065 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 917012480
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.16.0
|  python:
|    pip: 19.0.1
|    version: 2.7.13
|    virtualenv: /home/pi/oprint

derpicknicker1, 2019-08-08:
I have this problem too on one of my Octoprint instances. This has to be investigated.

m-ahrens, 2019-11-11:
I get this exception as well. This holds me from starting a print remotely because I can not select a file. 
What's the state of this issue? What can I do to support fixing this?

m-ahrens, 2019-11-11:
I've played around with the files a bit and made the following observations:
Obviously the plugin cannot open a folder if the .metadata.json file does not contain a hash attribute for at least one of the files in the folder. If the .metadata.json is not present or every JSON entry has a hash attribute everything works as expected. 
It seems like that the fileList method relies on the existence of this attribute, but obviously OctoPrint does not necessarily create this....

steve2211, 2019-12-17:
Still having the same problem. Is there a solution yet?

juguerin3, 2019-12-18:
hi, sorry i dont' find any solution. The only one look work is the complete reset and reinstall the raspberry.

steve2211, 2020-01-25:
Unfortunately reinstalling the pi didn't bring a solution either...

jacopotediosi, 2025-07-10:
Two things to improve here:
- check if there is something we can do even when the hash is missing
- if a hash is missing, do not crash; continue serving files and folders that have a hash by handling the exception for the individual file rather than for the entire files list

jacopotediosi, 2025-09-02:
In the end I fell down the rabbit hole and decided to completely rewrite the logic of the `/files` command.
I'm fairly confident that commit d22cb9b2b34608de6293d1e0436bd05c405a9c96 introduces several improvements in handling edge cases that could previously cause crashes.

closed 2025-09-02 by jacopotediosi