jacopotediosi/OctoPrint-Telegram #194: Status only working once

patschi, 2019-08-08:
I'm having a very strange issue:. For whatever reason the `/status` command only works once - as seen here:
[image]

The bot appears to work, however not reacting to the status command anymore.

It is reproducible following way:
1. Go to "Zchange"
2. Set "Command message" to "Separated"
3. Set "Markup selection" to "MD" (Markdown)
4. Save settings
5. Now `/status` only works once.
6. The even funnier thing: When you go to the same settings and you change back "Command message" to "Combined", it still doesn't work once and there's no message appearing anymore.

tl;dr: Something when the Markdown parser or so seems to be broken.

giloser, 2019-08-08:
Hi 

did you have the gif activate? it can take some time before he respond, we will add a temp message to let know that the gif is in creation .

patschi, 2019-08-08:
No, I haven't enabled the new GIF feature on that message. As written earlier, once I change "Markup selection" to something else than "MD", it works instantly just fine. (also without GIF)

Not related to the issue: I tried using `/gif` before once, but then nearly killed the printing job and the whole Raspberry Pi went OOM. I believe generating a GIF from multiple 3240x2464 photos simply takes too much resources from that little thing.

giloser, 2019-08-09:
Hi,
as I have made an error and the gif is activate by default for the Z change message can you check your the gif features is really not activate for this message?

thanks

patschi, 2019-08-09:
In the main settings above, it is:
[image]

But next to the Zchange it isn't:
[image]

jacopotediosi, 2025-07-10:
The user did not attach their logs. However, it is likely something related to webcam handling or message formatting, which should have been fixed by recent commits in the `devel` branch (e.g., 1dd2fd61b0eb8edec72fd2e4c4970fedd24da8cf).

closed 2025-07-10 by jacopotediosi