jacopotediosi/OctoPrint-Telegram #197: GIF feature nearly killing RPi

patschi, 2019-08-10:
While I really like the new idea of the new `/gif` feature I believe it might still need a few optimizations. I do use the official Raspberry Pi camera v2 and have configured the camera at mode 2 `3240x2464 @ 15 FPS`.

When I do use `/gif`, the result in `htop` on the Raspberry Pi is following:
[image]

Without the Raspberry Pi 4 with 4 GB of RAM, it might also caused some more serious issues like killing processes as it ran out of RAM. As older RPi versions do just have 1 GB of RAM.

Furthermore: I had to wait about 2 minutes to receive a nearly 12 MB large-MP4 which was about 3-seconds long. Unfortunately three seconds isn't quite long, you basically can't see a lot in that timeframe. I'd suggest recording (if possible) a pre-defined amount of seconds (e.g. 7 seconds) at a fixed quality (e.g. 720p) and then send the MP4 back.

I know, it might be related due to the nature how large that photos are and how the whole process works underneath. I just wanted to report my experience using the GIF feature. Unfortunately in the current state I can't really use the GIF feature overall, as it noticeably impacts the printing process (even more on the 3B+).

marksie1988, 2019-08-11:
ah yes im having this same issue, it disconnects my printer and causes prints to fail.

giloser, 2019-08-15:
Hi,

sorry to hear that it kill your print.
It's true I use it on a PI3 with a pretty bad camera.
We'll try to force it to use less CPU on the gif creation.
unfontunnately for the moment I have'nt found a way to get acces to the webcam directly so I ask Octoprint to send me an image every .5 seconds. I haven't found a way to set the quality to octoprint so the files have the quality you set on octoprint.

So next release will add a way to use less CPU, unactivate by default and even add a checkbox to activate or desactivate all the gif features.

I'll see if I can add option on the screen to let people decide how many images and the time between image, scale option to try to reduce size , ...

patschi, 2019-08-21:
I just come across an other idea: Octoprint does have a setting for a "Stream URL" in the settings, which is `/webcam/?action=stream` by default. This is a video stream. You might be able to record a pre-defined amount of seconds of this stream at a lower resolution (e.g. 720p, way enough to see most important parts), convert it over to GIF and then send that back to the user.

That might be a better approach than always requesting a screenshot every X seconds, wouldn't it?

giloser, 2019-08-22:
Hi

I try that road also and I would love to use that way but that means install opencv and I haven't found a way to install it on an easy way on a raspberry.
so if someone has a nice solution for that, that we can be sure will work for everyone I will be glad to try.

rlogiacco, 2019-08-23:
I fail to understand why is opencv required... but I admit I’m no expert in the field

giloser, 2019-08-24:
hi 

I search more and found a way to catch the webcam as a stream directly from the URL with ffmpeg but we catch what is defined in octoprint

so mine in 640x480 give a file of +-6mb for 5sec of video so not usable

if I tried to scale the output video or reduce the fps I can lower the size of the video but then again the cpu is highly loaded.

I am frankly out of idea right now if anyone has another one?

thanks

giloser, 2019-08-29:
Hi,

I finally made some change to try to really fix this one and for all
the version is in a pull request but if someone want to try it here is should be the link to install directly on the plugin manager
https://github.com/giloser/OctoPrint-Telegram/archive/master.zip

adrensnyder, 2020-01-08:
Hi all.

I have the same problem and don't know how to try the zip. I have only to unzip where octo install the actual plugin? Meybe someone already tried?

Maybe this zip is too old?

Thanks

adrensnyder, 2020-01-09:
> Hi,
> 
> I finally made some change to try to really fix this one and for all
> the version is in a pull request but if someone want to try it here is should be the link to install directly on the plugin manager
> https://github.com/giloser/OctoPrint-Telegram/archive/master.zip

I tried installing the patch and cpulimit on my raspbian and now the cpu is less used. 
The system is an Asus d250 netbook (old with only 2 core and 1,5ghz cpu) and the prints do not stop even if I hope that you will find another less heavy mode to acquire it. 
Anyway great plugin and thanks

Zogar89, 2020-01-30:
Gif and LargeGif feature are killing the CPU and make layershifts. Raspberry Pi3+ 1.5.0 release.

[image]

closed 2021-02-09 by giloser

jordansilva, 2022-01-05:
I don't know why this issue was closed. I tested supergif today on RPi4 and no video received and I lost the remote access to octoprint. My print is still running though.