jacopotediosi/OctoPrint-Telegram #201: Filamentmanager + äöüß in Filament-Name = Error

Peter-J, 2019-08-14:
TL;DR: Plugin has problems handling "ä ö ü ß" and similar symbols in a filament-name.

-------- Long version -----
i got: 😖 Error getting spools. Are you sure, you have installed the Filament Manager Plugin?

debug log: (i removed most spools from the log)

> 2019-08-14 18:32:22,721 - octoprint.plugins.telegram.listener - INFO - Got a command: '/filament' with parameter: 'spools' in chat *****
> 2019-08-14 18:32:22,727 - octoprint.plugins.telegram.TCMD - INFO - Parameter received for filament: spools
> 2019-08-14 18:32:27,174 - octoprint.plugins.telegram.TCMD - INFO - Spools: [{u'profile': {u'diameter': 1.72, u'material': u'PLA', u'vendor': u'Motedis', u'id': 3, u'density': 1.25}, u'temp_offset': 0, u'name': u'**Gr\xfcn**', u'weight': 330.0, u'used': 11.471861346984554, u'cost': 5.4, u'id': 6},{u'profile': {u'diameter': 1.74, u'material': u'PETG', u'vendor': u'Motedis', u'id': 6, u'density': 1.38}, u'temp_offset': 0, u'name': u'**Wei\xdf**', u'weight': 330.0, u'used': 0.0, u'cost': 5.4, u'id': 21}]
> 2019-08-14 18:32:27,188 - octoprint.plugins.telegram - DEBUG - Sending a message UPDATE: 😖 Error getting spools. Are you sure, you have installed the Filament Manager Plugin? chatID= *****

After Changing "Grün" to "Gruen" and "Weiß" zu "Weiss" --> No error, works as intended.

rlogiacco, 2019-08-14:
Sadly it makes a lot of sense: the whole plugin has an issue with character encoding.

Peter-J, 2019-08-14:
I think one could forgo a fix (for the time being) and instead have a note somewhere saying something like "Avoid äöüß and other special characters - Applys to connected plugins too."

IMHO: poeple who use a 3D-Printer and are willing to setup octopi + telegram-plugin, are also willing to live without special-characters.

PR #202 (2019-09-09): Improve Unicode support
This pull request should fix #201 and other similar issues by switching from `str()` to `unicode()` for encoding strings sent via Telegram.

These changes have not been tested and other than filament manager integration (`/filament`) it also covers custom commands (`/cmd`): please test before releasing.
closed 2019-09-09 by giloser