jacopotediosi/OctoPrint-Telegram #205: Plugin never times out HTTP connection to get gif/screenshot

dzervas, 2019-09-12:
I found out about this through trial & error:

mjpg-streamer fails but the server is up (8080) but when you request a screenshot/stream it keeps the connection without ever timing out.
Then when a /status command is sent to the plugin, it never answers (to any command) until the connection is somehow closed (reboot/kill the server) where all the answers are sent in bulk with image error (as expected)

I suggest timing out the http on around 15s and letting the user know via telegram (maybe even an option to disable webcam until reboot?)

FranciscodeMaussion, 2019-09-13:
Same here!! I have a phone acting as an ipcamera and I make requests over HTTP, when sth goes wrong the telegram bot breaks, until I repair the camera server.

dzervas, 2019-09-18:
BTW why don't you run octoprint ON the phone?

FranciscodeMaussion, 2019-09-18:
Ididnt know I can do that!!!! How?

giloser, 2019-09-25:
Hi,
I don't know how I can reproduce such a behavior. I tried by killing the mpej-streamer but I got an answer directly without image.
Anyway I see where we could put the timeout and I don't think it will hurt so I will add it to my version, let it be for some days and push it to the version if I don't detect any trouble.

closed 2021-02-09 by giloser

dzervas, 2021-02-09:
I haven't had the bug for 2 years now but the culprit is mjpeg-streamer I think. I'm not sure but I think that mjpeg loaded forever.

Just close it till someone else reopens it