jacopotediosi/OctoPrint-Telegram #207: Many request to api.telegram.org

slipx06, 2019-10-06:
This plugin makes multiple requests to api.telegram.org. Seems to be every 30 seconds. I'm not sure if this is "normal" behavior. As a work around I created a manual entry in the /etc/hosts file to stop the constant lookups hammering my DNS server 

[image]

[image]

[image]

rlogiacco, 2019-10-06:
I’m not sure I understand where the bug is. Yes, it is normal for a telegram bot to poll for new messages, once every 30 seconds seems pretty fair compromise between responsiveness and load.
If a DNS can’t handle a simple cacheable request every 30 seconds I would be concerned by the DNS server capabilities rather than by the rightfulness of the requests....
Your DNS should receive the request and just reply with the same value you entered in the /et/hosts for around 24 hours, then it should check again if the value it’s still correct... you should have absolutely no need for hard coding it in your hosts file

slipx06, 2019-10-07:
Can we not optimise this i.e. don't poll if OctoPrint is offline?

rlogiacco, 2019-10-07:
If by _offline_ you mean **power down** than no request is issued by a device that is powered off, if you mean _disconnected from the internet_ then the only way to know that is by submitting a request...

If you mean _reduce the poll frequency when multiple "unavailable host" responses are returned_, feel free to submit a pull request implementing such a feature: the plugin author is generally welcoming any improvement.

Personally I don't feel the need for that: I'm generally connected to the internet and my 20€ home router happily complies with the current poll frequency.

slipx06, 2019-10-07:
When I say offline, I am referring to the printer being offline or disconnected. As I understand this plugin is only used when the printer is online.

MarcusWolschon, 2020-02-21:
No, the plugin works fine when the printer is not connected.
That's what the /con command is for.

jacopotediosi, 2025-06-25:
This behavior depends on the DNS TTL of Telegram's Bot API.

Of course I could add a setting to ignore the TTL and cache the IP address for a user-configurable amount of time, but I'm not sure why Telegram sets such a low TTL, and I can't predict what side effects this might have.

As far as I can tell, the current TTL for `api.telegram.org` ranges between 30 seconds and 5 minutes. These values don't seem low enough to justify a workaround, so I'm closing this issue.

I'm open to reconsidering if you can provide a use case where this behavior has a real impact on your setup.  
Keep in mind that you can still cache the IP at the operating system level where OctoPrint is installed, or via your local DNS server.

closed 2025-06-25 by jacopotediosi