jacopotediosi/OctoPrint-Telegram #210: Percentage sent is different from the one in OctoPrint

soemarko, 2019-10-20:
I just noticed this after using this plugin for awhile. It sent the message that the print is 84% done while the one on OctoPrint showed that it's still at 74%. But the print time and ETA time is the same.

Not a big deal, just a little annoying.

peskyfish, 2020-11-20:
yep, got the same issue.
Octoprint showing: 51% (layer 40/199)
Telegram Notification showing: 39%
Time: 5:23 done, 5:07 remaining

jacopotediosi, 2025-07-19:
I wasn’t able to reproduce this bug, and I’m fairly confident it was fixed in the past—though I’m not sure exactly when.

Specifically, looking at the code of the `_sendNotification` method, which is responsible for replacing variables in notification messages:

https://github.com/jacopotediosi/OctoPrint-Telegram/blob/857f4353bfdb0115f8ee267069bdcbe4c42f4616/octoprint_telegram/telegram_notifications.py#L295  
https://github.com/jacopotediosi/OctoPrint-Telegram/blob/857f4353bfdb0115f8ee267069bdcbe4c42f4616/octoprint_telegram/telegram_notifications.py#L327-L330

The `percent` variable is based on `progress.completion`, returned by OctoPrint’s `get_current_data` method, so it shouldn’t differ from what’s shown in the OctoPrint UI.

If this issue still occurs, please open a new issue and include the G-code file you’re printing, the logs requested by the issue template, a list of installed plugins, and a screenshot highlighting the discrepancy between the percentage values.

closed 2025-07-19 by jacopotediosi