jacopotediosi/OctoPrint-Telegram #211: Change the timezone

Kawuezel, 2019-10-21:
Hi,

I have the issue that in Telegram, the plugin is sending me the false ETA time. It's 2 hours behind my timezone in central europe. On the Octoprint dashboard, it's showing for example 15h18, but in the message in telegram, it shows 13h18.

Is there a way to change the timezone, or is there another problem?

Thanks,
Kim

giloser, 2019-10-21:
Hi ,

have you check/configure the timezone of your PC/raspberry
on a raspberry it could be done with the raspiconfig.

Gilles

MarcusWolschon, 2019-10-22:
Given that Octoprint uses the same timezone, this seems unlikely.

w4-sglim, 2019-11-11:
I just changed the timezone in Raspberry pi so I'm sure that octoprint-telegram uses the same timezone with octoprint and raspberry itself.

Kampfqualle, 2019-12-17:
I have the same issue.
Time in OctoPrint is correct, but Telegram is 1h behind.

melvinisken, 2020-03-25:
Same for me. Telegram is 1h behind.

ftzhurrey, 2020-04-02:
Yeah, got the same problem. 1h behind.

developnoop, 2020-08-07:
Same Problem here, central europe. 
Do you need any kind of log file or other info?
The ETA you see in the GUI of Octoprint is correct, but the Telegram ETA is 1h behind.

SM-26, 2020-09-25:
Still got the problem.
If someone found a workaround, please let us know

texttoolong, 2020-10-23:
Please SSH into your octopi.local.
Type "date" - you will probably get returned the wrong time (unset = set to London time). 
Type "sudo dpkg-reconfigure tzdata" and select your time zone. 
Done.
All the best, Thomas

developnoop, 2020-10-23:
> 
> 
> Please SSH into your octopi.local.
> Type "date" - you will probably get returned the wrong time (unset = set to London time).
> Type "sudo dpkg-reconfigure tzdata" and select your time zone.
> Done.
> All the best, Thomas

You are 100 % correct. This solved the issue!
Maybe someone should add this to the wiki here?
@texttoolong Thank you for this insight!

closed 2021-02-09 by giloser

LeightonSolo, 2024-10-20:
I'm using Octo4a on an android phone as my Octoprint installation. The phone timezone is correct but it isn't in Octoprint, any idea how to change for Android instead of raspberry pi?