jacopotediosi/OctoPrint-Telegram #212: Exceptions before every send message

TheKrush, 2019-10-31:
```
2019-10-30 20:55:42,326 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: str
2019-10-30 20:55:42,334 - octoprint.plugins.telegram - INFO - Sending a message: Started printing [x4]Raging.gcode. with_image=True with_gif=False chatID= 512061463
```

Been seeing this in my logs.

TheKrush, 2019-10-31:
Messages do get sent though.

poikilos, 2020-01-25:
I am getting this too: `2020-01-24 23:24:28,560 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: str`

giloser, 2021-02-09:
is this still a problem in the latest version?

thanks

jacopotediosi, 2025-06-24:
Duplicate of #257.  
Closing this issue in favor of the other one since it contains more details.

closed 2025-06-24 by jacopotediosi