jacopotediosi/OctoPrint-Telegram #213: [Request] Notification on M600

MaxM4X, 2019-11-03:
It would be great to get a notification for M600 so you know when you have to change the filament on multicolor prints or when the filament runs out.

Quintin9801, 2019-11-11:
This feature would be a lifesaver!

roboraptor, 2021-01-04:
I would LOVE that!

Kesiunas, 2021-01-07:
Would really love that!! Installed this plugin just for this kind of feature:)

giloser, 2021-01-07:
Hi there was a way that already exist and it's to add the https://plugins.octoprint.org/plugins/PauseForUserEvent/

it was sending a message when the printer was waiting for a user interaction but it seems to be stuck in python 2 and nothing move from some times.

So I try to reproduce what he was doing inside the plugin directly.
I will post a test version here in the next few days with a specific message for a M600 gcode and a PauseForUserEvent integrate in the plugin.

Kesiunas, 2021-01-07:
> Hi there was a way that already exist and it's to add the https://plugins.octoprint.org/plugins/PauseForUserEvent/
>
> it was sending a message when the printer was waiting for a user interaction but [...]

Nice! Can't wait to test it out. Thank you very much

giloser, 2021-01-08:
Hi,

here is a test version for a M600 message

https://github.com/giloser/OctoPrint-Telegram/archive/TEST_M600.zip
You can install with octorprint in the plugin manager there is an option to install directly from link

don't forget to activate the notification for the new message in the plugin options

[image]

baeriwiliwonka, 2021-01-10:
Great, exactly what I am searching for.
But the installation does not work on my side.

`Installiere Plugin von https://github.com/giloser/OctoPrint-Telegram/archive/TEST_M600.zip...`
That's it... nothing happens

giloser, 2021-01-10:
That's strange I just tried again and it installed without problem on mine. Could you try again and provide the log file?

until now on python 2 the plugin_pause send a message when he receive on the terminal a specific message "echo:busy: paused for user"
And I have seen that this message only appears when the M600 finished the unload. So I decide that it would be better if we separate the 2 and we get a M600 message as soon as it start so we have time to get to the printer.

baeriwiliwonka, 2021-01-10:
[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5792757/octoprint.log)

Attached the logs.
Trying to install with link: Stuck at `Installiere Plugin von https://github.com/giloser/OctoPrint-Telegram/archive/TEST_M600.zip...`
Installing from file: Unknown error

giloser, 2021-01-10:
Ok sorry I need the ~/.octoprint/logs/plugin_pluginmanager_console.log to see something about the installation.

but maybe you can try first with the downloaded zip
[OctoPrint-Telegram-TEST_M600.zip](https://github.com/fabianonline/OctoPrint-Telegram/files/5792809/OctoPrint-Telegram-TEST_M600.zip)

thanks

baeriwiliwonka, 2021-01-10:
I already tried by installing the downloaded file, result was unknown error, see post before.
But it seems that I have a general issue, since other Plugins are also not installing.
I recognized that also an update of my firmwareupdater did not work, I got an errormessage that the update was not started and an information like "is another update running?" - but it is not.

the plugin_pluginmanager_console.log shows no current data, the file is old.

Kesiunas, 2021-01-10:
It installed correctly for me. No problems at all. I haven't had some time to test it though. I will get back as soon as I will have the feedback. Thanks for the update

baeriwiliwonka, 2021-01-11:
You are right it is worling fine!
I had some general problems with my octopi, but now everything is fixed and I was able to install your test version.

Only thing I recognized is that I got the message twice: First when the filamentchange layer was reached and second when I load the new filament. And the default message conatins "second try" which makes no sense for me, but this is editable easily.

So beside the doubled message this is a very nice solution!

Thank you very much and sorry again for the confusion based on my faulty octopi installation.

giloser, 2021-01-11:
Hi happy to hear you have been able to install.

yeah there is 2 different message because the M600 come sooner that the pause for user and yes the pasue for user would have worked before if you had the plugin installed, but now everything is directly inside this plugin so the other one is no more needed as it seemed abandoned.

baeriwiliwonka, 2021-01-11:
Will this find the way to the stable release of the telegram plugin?

giloser, 2021-01-11:
yes I'll wait some days to be sure it work as expected for the others as well and then I'll create a new version.

Kesiunas, 2021-01-14:
Hello,

I've tested this plugin addon and it works perfectly as expected!
At the same time, pause for user interaction started to work because before it was not sending me any notifications that printer is waiting.

Thank you very much it makes my work much faster

closed 2021-02-09 by giloser