jacopotediosi/OctoPrint-Telegram #214: [Request] Multiple Camera support

akargl, 2019-11-04:
It would be useful if one could configure several camera (urls) and get sent different perspectives on each step instead of just the default cam of Octoprint.

SchrauberGuru, 2019-11-04:
Thumbs Up... :-)

i would also appriciate to get Informations about all printers. because i´ll get up to 8 printers working.
Your problem will be to define instance names which will be defined by your haproxy conf.
maybe its possible to enter own defind URLs further. meanwhile status/proceeding from all working printers would be enough for me.

BR,

TheAndi, 2019-11-09:
It seems to be pretty easy to adapt to multiple cameras, as can be seen here:

https://www.reddit.com/r/octoprint/comments/9el0pg/octoprint_telegram_plugin_modified_to_send/

It would be great to get an official option in the plugin to do that!

giloser, 2019-11-12:
Hi,

yes except it's not that simple you would expect this plugin to be able to get the multicam options and not to have to configure it twice.
So we could check if the plugin is there , then we have to found how the multicam options are stocked and access them and then we could loop on get images or gif from the different sources.

I just installed the multicam and configure it to work with 2 cameras but for now I haven't been able to found the options in a way that will not force us to configure it twice and will not break the app for all the users that doesn't use the multicam.

Gilles

TheAndi, 2019-11-15:
I think most of the users with multiple cams wouldn't mind setting it up twice instead of not having this functionality. In every other plugin we also have to configure it twice. I think an interface from octoprint itself for more cameras would be the best solution. But until then it would still be nice to be able to configure it by hand.

giloser, 2019-11-15:
well since I install it, I got some times to check 
and I have found a way to get the config from the multicam.
so for me it's now working for the gif. I still have some trouble with the other image I will let you know when I have a version to test.

Astrofreak85, 2019-12-12:
Hi, I would like to have that option too, Since I have a secound Cam wich looks at my filament ;)

Sadly the stream is from motion, an motion only provides an URL like this: http://172.16.1.2:8081/
Maybe I find an other (better) way to access this webcam (wich is on an other divice, not the one runnig Octopi....

Kampfqualle, 2020-01-28:
Would it be possible, to select a other stream with the /gif /supergif manually?
Like „/gif 192.168.178.XX:8083“ for the 3rd Webcam installed?

closed 2021-02-09 by giloser