jacopotediosi/OctoPrint-Telegram #218: Gif not loading in Telegram app....

mygitcrazy, 2019-11-11:
Everything is working as expected and i really like it. Only problem is that the gif (cam snapshot) that's being send is showing up, but stays blurry with a spinning wheel in the middle...

What could be the problem? I've tried different settings in the snapshot url settings and played around with the notification settings but no luck :-(

mygitcrazy, 2019-11-11:
[image]

mygitcrazy, 2019-11-11:
[image]

giloser, 2019-11-12:
Hi there,

it's seems that the screen shot from connection has the same problem so It's probably not a GIF issue.
Are you in Wifi or in data connection on your device with telegram?
how are your settings in the Data and storage?
does the Telegram app has the right to save local data on your device?

thanks
Gilles

mygitcrazy, 2019-11-12:
Thanks for the good advice ;-)) 

I checked the app settings, Data and Storage, that one was checked and active. That wasn't the problem. The problem is in the Wifi and data connection. As soon as I switched over from Wifi to Mobile Data connection the snapshots (gifs) came through. But one question left... why is it only available through Mobile Data connection? .......

mygitcrazy, 2019-11-12:
Ps

I shall close this issue as soon as possible.. cause it solved my problem till so far and I'm happy it works now :)

mygitcrazy, 2019-11-13:
One question left... why is it only available through Mobile Data connection? .......

giloser, 2019-11-13:
well it's working fine for me on wifi or mobile data.

Could you show a print screen of your telegram settings for that part?

thanks

mygitcrazy, 2019-11-13:
That's weird for my case then. Could it be a firewall related thing maybe? My network is in a heavily protected environment. What kind of settings of  Telegram do you need by the way? I see a lot of settings over there....

Thanks for your help by the way :-))

closed 2019-11-17 by mygitcrazy