jacopotediosi/OctoPrint-Telegram #233: Profile pictures not being displayed when using reverse proxy with subdir

CodeCrafter912, 2020-02-11:
Hey,

I have my Octoprint instance on "mydomain.com/octoprint/" but the profile pictures in the "known chats" list point to "mydomain.com/plugin/..." and can therefore not be loaded.
I know, that's a minor bug but it would still be nice to see it being fixed.

Best Regards and thanks in advance
Dorian

MarcusWolschon, 2020-02-12:
I uses the Apache mod_substitute in my reverse proxy to rewrite URLs contained in the html.
Works find even with very complex websites.
https://httpd.apache.org/docs/2.4/mod/mod_substitute.html

rlogiacco, 2020-02-13:
Calling this a _bug_ is a mistake: you have personalized your Octoprint installation using an additional frontend (Apache HTTPD) which is mangling with both the requests and the responses to and from your Octoprint installation. How can you think the Telegram plugin could be aware of your personalization?
If you want such a configuration your option is to fork the project, jump into the code and replace every point where the plugin returns an URL with your personalized version (which, BTW, is exactly what mod_substitute does for you on every HTTP response)

rlogiacco, 2020-02-13:
For your knowledge:

> `mod_substitute` provides a mechanism to perform both regular expression and fixed string substitutions on *response bodies*.

MarcusWolschon, 2020-02-13:
> It’s a very heavyweight mod to utilize, I hope you have plenty of CPU power because a complex website can be significantly slowed down by such a plugin.

That discussion would derail an already off-topic bug report.
It's fine as a workaround for single-user, low volume use cases similar the one suggested here.
Indeed, it's not a good choice for heavy-load websites with many users.

I suggest to close this issue. It has been triaged and a workaround for the original poster has been provided. Everything else CodeCrafter912 should be able to figure out on his/her own.

CodeCrafter912, 2020-02-13:
Thanks for the advice!
The mod_substitude works fine for me :)
If anybody else runs into this issue:
I added the following lines to the configuration of the reverse proxy:
` 
AddOutputFilterByType SUBSTITUTE text/html
`
`
Substitute "s|/plugin/telegram/img/user/|/octoprint/plugin/telegram/img/user/|i"
`

closed 2020-02-13 by CodeCrafter912