jacopotediosi/OctoPrint-Telegram #236: Unable to start print with /print

Philipp92, 2020-02-23:
Since a few days I am not able to start a print with the command /print anymore.

After typing in /print I can browse in my folders and see every file, but I can not choose any of them.

Can anyone help me and know how to fix it?

I am using the version 1.5.0

rlogiacco, 2020-04-08:
I'm running Octopi 0.16 with Python 2.7 and Telegam plugin 1.5.0 and I'm experiencing the very same issue.
Looking in the logs I see:
```
2020-04-08 07:53:38,117 - octoprint.plugins.telegram.listener - DEBUG - MESSAGE: {u'update_id': 897870058, u'callback_query': {u'data': u'/print_5d7eec19|0|372926c60f288b4849db96b2e248345e', u'message': {u'from': {u'username': u'#####', u'first_name': u'OctoPrint', u'is_bot': True, u'id': ######}, u'edit_date': 1586328791, u'text': u'\U0001f4be Files in /covid-19    [1/1]', u'entities': [{u'length': 6, u'type': u'bot_command', u'offset': 12}, {u'length': 6, u'type': u'bold', u'offset': 12}, {u'length': 3, u'type': u'bold', u'offset': 18}], u'chat': {u'username': u'######', u'first_name': u'Roberto', u'last_name': u'Lo Giacco', u'type': u'private', u'id': #######}, u'date': 1586328783, u'message_id': 3058, u'reply_markup': {u'inline_keyboard': [[{u'text': u'\U0001f4c4 face-mask-clip', u'callback_data': u'/print_5d7eec19|0|084cff68ad82513d933d3381fbedd509'}, {u'text': u'\U0001f4c4 face-mask-clip-x3', u'callback_data': u'/print_5d7eec19|0|94b3c6e94804acf27bf3ffbd7673ec70'}], [{u'text': u'\U0001f4c4 face-screen-A4', u'callback_data': u'/print_5d7eec19|0|3fc1a95dda7d1777d4b4cba155354212'}, {u'text': u'\U0001f4c4 face-screen-A4-stackable', u'callback_data': u'/print_5d7eec19|0|cd4d8ea1912524412ecfff672120cf66'}], [{u'text': u'\U0001f4c4 maschera.pla', u'callback_data': u'/print_5d7eec19|0|624606483f30562b58e82931f3cda0f6'}, {u'text': u'\U0001f4c4 maschera_griglie', u'callback_data': u'/print_5d7eec19|0|372926c60f288b4849db96b2e248345e'}], [{u'text': u'\U0001f4c4 mask-screw.petg', u'callback_data': u'/print_5d7eec19|0|87324071d591a23e2f620e1598b64bea'}], [{u'text': u'\u21a9\ufe0f Back', u'callback_data': u'/print_0bca257b|0'}, {u'text': u'\U0001f4dd', u'callback_data': u'/print_5d7eec19|0|0|s'}, {u'text': u'\u274c Close', u'callback_data': u'No'}]]}}, u'from': {u'username': u'######', u'first_name': u'Roberto', u'last_name': u'Lo Giacco', u'is_bot': False, u'language_code': u'it', u'id': #######}, u'id': u'###########', u'chat_instance': u'##########'}}
2020-04-08 07:53:38,119 - octoprint.plugins.telegram.listener - DEBUG - Not changing update_offset - otherwise would reduce it from 897870059 to 897870059
2020-04-08 07:53:38,123 - octoprint.plugins.telegram.listener - INFO - Got a command: '/print' with parameter: '5d7eec19|0|372926c60f288b4849db96b2e248345e' in chat ##########
2020-04-08 07:53:39,474 - octoprint.plugins.telegram.listener - ERROR - Exception caught! 'hash'
```

rlogiacco, 2020-04-08:
I've investigated the issue a little more and it seems related to the nesting of the print file inside one or more folders: printing a file from the root folder works as expected, but if the file is inside a folder then the message above is produced.

derpicknicker1, 2020-04-08:
Known bug. Will be fixed with next release. Working on this.

Marlerman, 2020-04-28:
I have this same issue as well. I just uploaded a file to the root folder and that did not work either. Any luck on solving this issue? Thanks for working on it!!

[octoprint (1).log](https://github.com/fabianonline/OctoPrint-Telegram/files/4548509/octoprint.1.log)

giloser, 2021-02-09:
is this still a problem with the latest version?

thanks

closed 2025-06-24 by jacopotediosi