jacopotediosi/OctoPrint-Telegram #241: Octoprint 1.3.12

lackdaz, 2020-03-23:
:warning: ERROR: I was not able to format the Notification for 'ZChange' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'ZChange'.

```
Printing at Z={z}.
Bed {bed_temp}/{bed_target}, Extruder {e1_temp}/{e1_target}.
{time_done}, {percent}%% done, {time_left} remaining.
Completed time {time_finish}.
```
Just installed Octoprint-Telegram and I must say that its phenomenal! But what is this bug? I'm on pretty much mint settings.

giloser, 2021-02-09:
is this still a problem with the latest version?

thanks

jacopotediosi, 2025-07-02:
The user did not attach [the logs](https://github.com/jacopotediosi/OctoPrint-Telegram/wiki/Where-are-the-logs%3F), which is why I’m closing this issue.  
However, it was likely related to the lack of variable escaping, which should have been solved by commit `af5c04c99f4a0b8d018bf3d0943e6465aa995efe`.

closed 2025-07-02 by jacopotediosi