jacopotediosi/OctoPrint-Telegram #244: Incompatible Python 3.6

gvmturl, 2020-04-03:
Could you please update it?

Thank you for the job!

gvmturl, 2020-04-11:
Cant use the plugin with python 3.7 any solution please:

rlogiacco, 2020-04-11:
The author and his apprentices are working on this, you might want to find a way to support his efforts and show your appreciation for his donation to the community ;-)

gvmturl, 2020-04-20:
Tried adding __plugin_pythoncompat__ = ">=2.7,<4" on __init__.py of python3 branch. could make it work but the listener is not working... 
ERROR: I was not able to format the Notification for 'PrinterStart' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrinterStart'.

And 

I do not understand you! 

Are the messages that i m receiving from bot... any help?

Thank you guys!

gvmturl, 2020-05-01:
Is there any news about the python 3 compatibility?

szobov, 2020-07-16:
Hi there,
I've fixed **python3** incompatibility in my fork: https://github.com/szobov/OctoPrint-Telegram
But I didn't prepared it for this repo, because I didn't spend time to fix it for **python2**.
You're welcome to check:
```
python -m pip install "https://github.com/szobov/OctoPrint-Telegram/archive/master.zip"
```
Cheers.

MarcusWolschon, 2020-07-16:
Can you create a pull request?

szobov, 2020-07-16:
> Can you create a pull request?

Sure, here it is:
https://github.com/fabianonline/OctoPrint-Telegram/pull/273

xbarents, 2020-09-11:
Octoprint made a anncouncement for a conversion script to migratie from python 2 to 3. We can expect more questions coming because Telegram will not work anymore after running the script.

Is there any timeframe when the plugin will be updated?

maier-johno, 2020-09-16:
Hello, i have tested it too. First, the upgrade was without errors. But my telegram-plugin don't work. And my FilamentManager-Plugin can't connect to my database on a other RasPi. So, i go back with your " https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py" script. This works great too!

That's the only reason why i go back. But respect, both scripts works perfectly! Octoprint with Python 3.6 too!

Great job and thank's for the great scripts!

fractalsauce, 2020-09-20:
@Thanks for the fix @szobov :)

FranciscodeMaussion, 2020-09-24:
This is mandatory, for now we can use [szobov solution](https://github.com/fabianonline/OctoPrint-Telegram/issues/244#issuecomment-659382633). Thanks @szobov

PR #291 (2020-10-10): [Fix #244] Fix incompatibilities with python3
Closes #244

Should work for both **python2** and **python3** version.
But need to be tested on both.

To test this feature you can use this link to install using Plugin Manager `install from URL`:
https://github.com/szobov/OctoPrint-Telegram/archive/fix-244-python3-incompat.zip

Or install from terminal (but check you're using the same version with which Octoprint has been installed):
```
python -m pip install https://github.com/szobov/OctoPrint-Telegram/archive/fix-244-python3-incompat.zip
```

Prev PR: #273 

Thanks.
commit f083d3e6 (2020-10-10): Merge pull request #291 from szobov/fix-244-python3-incompat

[Fix #244] Fix incompatibilities with python3
closed 2020-10-10 by giloser