jacopotediosi/OctoPrint-Telegram #252: The notification settings doesn't work with Firefox

peppelorum, 2020-04-12:
The notification settings behaves very oddly when using Firefox.
You can use the settings and save, but if you reload the browser and view them again they are wiped. But with Chrome it works fine. If you change the settings in Chrome and then load the page in Firefox so you see the same settings, but If I change anything (and hit save) it doesn't get restored after a reload.

I have tested with a private window with all plugins turned off but no change in behaviour.

I'm using the latest Firefox (74) on MacOS Catalina.

derpicknicker1, 2020-04-12:
The issue only appears on telegram settings? Or do you see the same behavior on other settings?

peppelorum, 2020-04-12:
Just on the telegram settings, changing color of appearance etc works fine.

peppelorum, 2020-04-12:
If I check in config.yml I can see the right notifications being written, but as soon as I open telegram settings it gets restored to what previously was.