jacopotediosi/OctoPrint-Telegram #254: Enhancement: Wait for end script to complete before sending notification

KevinNovak, 2020-04-19:
It would be handy to wait delay the notification until the end script in Octoprint has completed.

For example, I would like to wait for the bed to completely cool before receiving the notification:
[image]

I know there is a delay option which is also handy but in my scenario the cooldown does not always take the same amount of time.

Regardless thank you for your work on this plugin, it's been incredibly helpful to me since my printer is multiple floors below me.

iaglet, 2020-05-14:
it might be the same thing but telegram is notifying me WAY before the print is complete that the print is complete.

sungo, 2020-05-21:
@iaglet I just filed a separate ticket for that behavior. I'm seeing the same thing. https://github.com/fabianonline/OctoPrint-Telegram/issues/262

closed 2021-02-09 by giloser