jacopotediosi/OctoPrint-Telegram #257: unsupported type for timedelta

mrseeker, 2020-04-24:
Hello,

Every time my print finishes the telegram plugin crashes octoprint with the following reason:

`2020-04-24 13:43:30,856 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: unicode
2020-04-24 13:43:30,865 - octoprint.plugins.telegram - INFO - Sending a message: Finished printing CE3_AIl_In_One_3D_printer_test_micro.gcode. with_image=True with_gif=False chatID= 570698329
`

Any idea why this is the case?

maniak26, 2020-05-01:
I got a similar error when I started using the IP camera.

`2020-05-01 16:42:41,879 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: unicode
2020-05-01 16:42:41,887 - octoprint.plugins.telegram - INFO - Sending a message: Finished printing Rod_Holder_Buttom.gcode. with_image=False with_gif=False chatID= 652437`

renatobo, 2021-02-05:
Same error still present, I also use an IP Camera

`2021-02-05 08:22:30,866 - octoprint.plugins.telegram - ERROR - An Exception in get final time : unsupported type for timedelta seconds component: str
Traceback (most recent call last):
  File "/Applications/Octoprint.app/venv/lib/python3.9/site-packages/octoprint_telegram/__init__.py", line 1465, in calculate_ETA
    finish_time = current_time + datetime.timedelta(0,printTime)
TypeError: unsupported type for timedelta seconds component: str`

giloser, 2021-02-06:
sorry about that I'll make sure to check about that for the next release

giloser, 2021-02-07:
when I check the number line I'm not on the same function as your log says. 
Are you sure you are on the last version?
thanks

renatobo, 2021-02-09:
That octoprint instance is no longer running, but it was as downloadable and with all updates from octoprint itself.

Same error on the new instance (OctoPrint Version 1.5.3 - Telegram Notifications 1.6.3) , different line
```
Feb  8 12:32:09 aspirone octoprint[933]: 2021-02-08 12:32:09,088 - octoprint.plugins.telegram - ERROR - An Exception in get final time : unsupported type for timedelta seconds component: str
Feb  8 12:32:09 aspirone octoprint[933]: Traceback (most recent call last):
Feb  8 12:32:09 aspirone octoprint[933]:   File "/home/octopi/OctoPrint/venv/lib/python3.6/site-packages/octoprint_telegram/__init__.py", line 1623, in calculate_ETA
Feb  8 12:32:09 aspirone octoprint[933]:     finish_time = current_time + datetime.timedelta(0, printTime)
Feb  8 12:32:09 aspirone octoprint[933]: TypeError: unsupported type for timedelta seconds component: str
```

Environment

- env.os.platform: linux
- env.python.pip: 21.0.1
- env.python.version: 3.6.9
- env.python.virtualenv: true
- octoprint.safe_mode: false
- octoprint.version: 1.5.3

These plugins might be relevant to Time

- ProgressBasedOntime Plugin (0.1.6)
- PrintTimeGenius Plugin (2.2.7)

giloser, 2021-02-09:
ok do you know if that happen on an event or when you select the files?
I don't seems to be able to reproduce that error .

renatobo, 2021-02-15:
It happens at the completion of a print, more context from the log itself - please let me know if there is debugging I can activate.
While this is a minor issue (I receive notifications and - it works great, thank you!) I am trying to get rid of error messages in log files.

```
Feb 15 01:35:57 octoserver octoprint[901]: 2021-02-15 01:35:57,275 - octoprint.plugins.detailedprogress - INFO - Message: ETA 2/15 1.38AM
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,178 - octoprint.util.comm - INFO - Finished in 8075.665 s.
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,179 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,198 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: spool_holder_side.gcode, owner: renatobo
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,391 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,394 - octoprint.plugins.filamentmanager - INFO - Updating Filament usage for octoprint configured toolcount: 1
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,395 - octoprint.plugins.filamentmanager - INFO - Filament tracked toolcount: 1
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,396 - octoprint.plugins.filamentmanager - INFO - Filament tracked values: [8219.182670000155]
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,403 - octoprint.plugins.filamentmanager - INFO - Filament used: 8219.182670000155 mm (tool0)
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,456 - octoprint.plugins.filamentmanager - INFO - Updated remaining filament on spool 'PETG - Black - PETG (SUNLU)' from 776.0429112309384g to 751.3310973474488g (-24.71181388348964g)
Feb 15 01:36:10 octoserver octoprint[901]: 2021-02-15 01:36:10,479 - octoprint.plugins.telegram - ERROR - An Exception in get final time : unsupported type for timedelta seconds component: str
Feb 15 01:36:10 octoserver octoprint[901]: Traceback (most recent call last):
Feb 15 01:36:10 octoserver octoprint[901]:   File "/home/pi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_telegram/__init__.py", line 1623, in calculate_ETA
Feb 15 01:36:10 octoserver octoprint[901]:     finish_time = current_time + datetime.timedelta(0, printTime)
Feb 15 01:36:10 octoserver octoprint[901]: TypeError: unsupported type for timedelta seconds component: str
```

I am also using the **DetailedProgress** plugin, which seems to suffer from a similar issue
```
Feb 15 09:26:19 octoserver octoprint[901]: 2021-02-15 09:26:19,587 - octoprint.plugins.detailedprogress - INFO - Message: ETA No ETA yet
Feb 15 09:26:19 octoserver octoprint[901]: 2021-02-15 09:26:19,589 - octoprint.plugins.detailedprogress - INFO - Caught an exception int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Feb 15 09:26:19 octoserver octoprint[901]: Traceback:Traceback (most recent call last):
Feb 15 09:26:19 octoserver octoprint[901]:   File "/home/pi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_detailedprogress/__init__.py", line 80, in do_work
Feb 15 09:26:19 octoserver octoprint[901]:     self._update_progress(currentData)
Feb 15 09:26:19 octoserver octoprint[901]:   File "/home/pi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_detailedprogress/__init__.py", line 86, in _update_progress
Feb 15 09:26:19 octoserver octoprint[901]:     progressPerc = int(currentData["progress"]["completion"])
Feb 15 09:26:19 octoserver octoprint[901]: TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```

giloser, 2021-02-16:
thanks for the details it's really helpful 
we get the info from octorprint and are expecting something like  printTime=13458.5947128 but we get printTime=11:11:00
for the example I have been able to reproduce.

I'll try to found something about that.

SnowViet, 2021-03-12:
I have the same problem:
`2021-03-12 19:20:38,838 - octoprint.plugins.telegram - ERROR - An Exception in get final time : unsupported type for timedelta seconds component: unicode
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 1678, in calculate_ETA
    finish_time = current_time + datetime.timedelta(0, printTime)
TypeError: unsupported type for timedelta seconds component: unicode`

commit 295243e1 (2025-06-27): Refactor calculate_ETA

Close #257

There should be (hopefully) no more crashes, because now in telegramCommands.py the operations that depend on printTime are only executed if it is defined.
commit 09929935 (2025-07-18): Refactor calculate_ETA

Close #257

There should be (hopefully) no more crashes, because now in telegramCommands.py the operations that depend on printTime are only executed if it is defined.