jacopotediosi/OctoPrint-Telegram #258: Notification settings an z-hop

Phil1988, 2020-04-26:
First of all: 
I just installed your plugin and its setup took like 5min. 
Thanks für the great manual, hints in the plugin itself (like the lists and documentation of how to insert emojis etc) and the overall great plugin!

First is, that I am using z-hop and have the notification set to every 5mm/15min.
But I am getting the notification on 5.0mm and (4.8mm or 5.2mm).
So every ZChange Message is send twice (overal print is less then 15min - so thats not the case).

How can I prevent this to happen?

rlogiacco, 2020-04-28:
I don't think you can, because your printer is crossing the 5.0 mm boundary multiple times... what about disabling notification on Z height and leaving only the time-based ones?

Phil1988, 2020-04-28:
That would be a workaround but using all features would be great ^^

Maybe its possible to include a flag that tracks the notification was already send.
Other then that a hysteresis would be usable as well to prevent this to happen.

I think many users out there are using z-hop so maybe something to think about :)

andyylin, 2021-06-16:
Yes, would be nice to fix this. It makes Z notifications too verbose, and I end up only using the time based notification

closed 2025-06-24 by jacopotediosi