jacopotediosi/OctoPrint-Telegram #260: Notification on height/time not working

olavgm, 2020-05-06:
I have 5mm and 30 minutes set for the ZChange notification. I don't get any. Commands work well (e.g. `/status` gives me a correct status with picture).

petersmythe, 2020-06-02:
+1, but I only enabled them after starting a print.  Perhaps they will take effect for a new print, but it would be great if they could take effect immediately.

FaFyle, 2021-02-26:
I have the exact same problem. Everything seem to work except this. I'm printing from octoprint, not from the SD card.

giloser, 2021-02-26:
so you have activate the notifications for your user?
can you share a log file please?

roboraptor, 2021-02-27:
What? I have opposite problem. I have octoprint set the same but it sends message every Zchange... 
I don't get it :D

giloser, 2021-02-28:
again could you share a log file with the debug option actif?

FaFyle, 2021-03-01:
Actually, after updating the plugin, it seems to be working fine now.

olavgm, 2021-03-01:
It's been working fine for me too for some time. Can't remember when I updated the plug in.