jacopotediosi/OctoPrint-Telegram #263: 3 Bugs since version 1.5.0: Status message, files and print command

hkleinsorge, 2020-05-22:
1. Status messages: I configured 10mm / 20min, but I mostly receive many every minute. Rarely, it works like configured.
2. /files command doesn't work. "Hash" error. But I found out, that there are gcodes it works with, and gcodes that dirsturb this function. I cannot see the difference in those gcodes. I can send you one of both kinds.
3. /print command doesn't work. Nothing happenes.

LeoS3, 2020-07-21:
I have two installations of Octopi+Telegram.. one works just fine, and the other exhibits the same issue as #1 above. The status notification seems to get triggered every minute, no matter what I set the height/time settings to.

Deses, 2020-10-11:
@hkleinsorge you are going to give more info if you want any sort of help with this.

Marlin version? Octoprint version? Plugin version? Files in the RPi or the SD? What board are you using? Anything that might help, put it here.
Also go ahead and attach those gcodes.

giloser, 2021-02-09:
is this still a problem?

jacopotediosi, 2025-07-10:
1. is a duplicate of #13
2. and 3. are duplicates of #193 

Also, next times please attach your logs

closed 2025-07-10 by jacopotediosi