jacopotediosi/OctoPrint-Telegram #266: Octoprint-Telegram connection middle-print

Satcar, 2020-06-17:
Hi,

So I just started working with this octoprint plugin and works perfectly. I managed to send the /print command from ~10km of my printer location, and was able to verify that it was printing with a webcam aswell.

The thing is that I arrived home and my phone automatically connected to my wifi network, and then the telegram bot stopped responding to my commands.

I was wondering if there's a any way to fix this.  I'm guessing that once it finishes the current print, I'll be able to reconnect to it again and the bot will start responding, but It'd be cool to be able to do that while printing.

Deses, 2020-10-11:
I see your problems are with WiFi, but does it work fine while using mobile data? Because if it works with mobile data, maybe there's an issue with your home network. Also try any other wifi network, just to see what happens.

What I've seen (using Telegram Desktop app) is that using `/con` and selecting AUTO connection mode, Telegram answers me with `Failed to start connection`, but it really does connect.

Satcar, 2020-12-04:
> I see your problems are with WiFi, but does it work fine while using mobile data? Because if it works with mobile data, maybe there's an issue with your home network. Also try any other wifi [...]

It works with mobile data, but only some times. I saw that the wifi signal I got on my printer was pretty bad, since It's relatively far away from the router. I bought a wifi range extender and it improved the wifi signal, but it didn't fix the problem.

My longest connection with the printer so far has been 2-3 hours. After that the telegram bot just stops responding/executing commands and I can't even reconnect, I have to reboot the device.

giloser, 2020-12-04:
When you have the problem do you can connect to the pi in ssh? with something like connectbot?
just to be sure it's not the pi that stop responding at all.

also can you tell the version of the plugin and octoprint please?

thanks

Satcar, 2020-12-04:
> When you have the problem do you can connect to the pi in ssh? with something like connectbot?
> just to be sure it's not the pi that stop responding at all.
> 
> also can you tell the version of the plugin and octoprint please?
> 
> thanks

Yes, I can connect to the raspberry using ssh since I used to reboot the device with it when I wasn't getting response.
I don't remember the version of octoprint, but it was the latest version available on June 2019. I gave up on this project on July but I'm willing to give it a second chance.

giloser, 2020-12-04:
No if the pi respond it should be ok from the PC, but we could confirm with a Desktop telegram on the PC if you receive the notification on this one and not on the phone then maybe the problem is more on the phone then in the plugin.
also we could if the problem is still there check the log to see if the plugin still log and try to send message or just stop to work.
I have also seen pi that seems to stop answering because of overload and start sending all the answer when the load is lower, did you use gif or multiple camera?
An another point we could check is the version of 'requests' I remember at some point the communication was really slow and it seems to come from that.
And a last point I'm thinking about is IPv4 and IPv6 , check if you have IPv6 active and try when it's not.

Hope this could help to find the issue.

Satcar, 2020-12-04:
> No if the pi respond it should be ok from the PC, but we could confirm with a Desktop telegram on the PC if you receive the notification on this one and not on the phone then maybe the problem is [...]

I tried deleting octoprint and performed a clean installation of everything , and It's been working for like 2-3 hours without any problem.
One thing I noticed is that I got a prompt after using the /gif command saying that I needed to install the "cpulimit" package. I don't remember installing it a few months ago.
I had issues with the pi cam as well now that you mention it... after like ~5 mins I remember getting black horizontal lines and eventually a white screen. I think I read that it was due current overload, since the cam draws a lot of power and the device gives up after a while... but I believe I had the gif option only, not the multiple camera. Maybe I ticked the multiple camera option as well and didn't notice it.

It seems like its working now! Thanks for the help
I'll post an update if I get any other issue.

closed 2025-07-10 by jacopotediosi