jacopotediosi/OctoPrint-Telegram #269: OctoPrint-Telegram fails with OctoPrint 1.4.1rc2

JohnOCFII, 2020-07-13:
Plug-in "disappears" with OctoPrint 1.4.1 Release Candidates.

I'm testing RC2, and my pre-existing installation of OctoPrint-Telegram no longer shows as installed.  I can install it again from the Plug-In selector, but after installation and re-start the plug-in again appears to be missing.  See also: https://github.com/OctoPrint/OctoPrint/issues/3638#issuecomment-657712892

Happy to provide any addition info/logs as requested.

cp2004, 2020-07-13:
Here's the reason it's failing: 
```Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 5, in <module>
    from flask.ext.babel import gettext
ImportError: No module named ext.babel
```

For the plugin author who fixes this, just change `from flask.ext.babel import` to `from flask_babel import`. This method has been deprecated, and removed in new versions of flask/babel, that OctoPrint 1.4.1 now depends on. (but flask-babel docs haven't been updated. See https://github.com/python-babel/flask-babel/issues/155)

markuznw, 2020-07-28:
Could you also fix the compatibility with python3? It installs with the PR but says incompatible... octoprint installed with python3

Thanks!

szobov, 2020-07-28:
@markuznw I took Gina's changes and pushed it in my fork's master. Please, check.
https://github.com/szobov/OctoPrint-Telegram/commit/e82f72ed2c76bf151af339d64f05f62177baf92c

I don't want to put it into  #273 because we already have #270 
Cheers

markuznw, 2020-07-28:
@szobov perfectly working, thank you for the super fast reply :)

szobov, 2020-07-28:
@markuznw just haven't disabled email notifications yet. :)
Thanks for testing, hope we'll see project's maintainers soon.

And thanks @foosel for the patch itself.

Crowlord, 2020-07-29:
@szobov being really dense here. How do i switch to your fork?

szobov, 2020-07-30:
@Crowlord you can find the instruction here: https://github.com/fabianonline/OctoPrint-Telegram/issues/244#issuecomment-659382633

Kedryn, 2020-07-30:
We really need a new release with this included 👍

giloser, 2020-07-30:
sorry but still need to be compatible with python 2.7 for all the people that have a working version.

foosel, 2020-07-30:
@giloser this has nothing to do with Python 2 or 3 and is perfectly compatible to both.

giloser, 2020-07-30:
ok then I'll check this again when I got the time.

foosel, 2020-07-30:
I've explained in #270 what the problem is (which btw has nothing to do with OctoPrint itself but rather a third party dependency). OctoPrint 1.4.1 release is imminent. If you don't merge #270 until early next week, this plugin will stop working for anyone who updates. I'd really appreciate it if you could save me the additional overload from disgruntled comments targeting me for something like that by merging this and pushing a new release BEFORE that happens, especially since this PR has been open for well over two weeks now and the RCs of 1.4.1 have been out for three 😕

szobov, 2020-07-30:
> sorry but still need to be compatible with python 2.7 for all the people that have a working version.

@giloser I suppose you're talking about changes in my MR #273 not about Gina's one. So, her changes do not break anything as I could say.

But the question is also about python3 support which I've introduced in #273 . AFAIK in the last releases of Octoprint they've dropped python2 support, so the plugin will not work already, and folks around this plugin were asking about python3 support.

If you want, we can prevent people, that use python2, from install new release. It can be resolved on the `setup.py` level, by putting this lines:
```
setup_parameters = octoprint_setuptools.create_plugin_setup_parameters(
python_requires='>=3',
....
)
```
You can find documentation [here](https://packaging.python.org/guides/distributing-packages-using-setuptools/#python-requires)

_disclaimer_
Yes, it won't work with old setuptools version, but it shouldn't be a big issue.
And I juse want to remind you that python2 is officially obsolete. 

Cheers.

foosel, 2020-07-30:
> AFAIK in the last releases of Octoprint they've dropped python2 support

Wrong. OctoPrint 1.4.x is perfectly compatible to both 2 **and** 3 and will continue to be. OctoPi 0.18 however will ship with Python 3.

I'd love to learn why people keep claiming that OctoPrint now only supports Python 3, you aren't the first and likely won't be the last. The release notes of 1.4.0 are pretty clear that it supports Python 3, not that it requires Python 3.

foosel, 2020-07-30:
Also, just for the record, "they" in reference to OctoPrint's developers is pretty much just me ;)

szobov, 2020-07-30:
> Also, just for the record, "they" in reference to OctoPrint's developers is pretty much just me ;)

Thanks for clarifications, Gina. I'm sorry, my bad, but I was pretty sure you've dropped python2's support.
And also sorry, I thought you aren't alone in maintaining such a big project., my respect :)

So, if someone wants he could use my changes and fix it to also python2 support. Unfortunately, I don't want to do it.

closed 2020-08-01 by giloser

foosel, 2020-08-03:
Closed but as of yet still unreleased 😕

PR #270 (2020-08-01): Fix use of deprecated flask.ext imports
`flask.ext` has been deprecated [for years now](https://github.com/pallets/flask/issues/1135) and should actually have been logging [a warning](https://github.com/pallets/flask/blob/0.12.x/flask/exthook.py#L69) to console ever since too. OctoPrint 1.4.1 had to update the flask dependencies due to security considerations and in the new flask version the old compatibility layer still allowing `flask.ext` to work got removed.

Which means this plugin no longer will load on OctoPrint 1.4.1+ unless this PR gets merged and a new release (1.5.1?) pushed out.

S [... cut]
commit ddcfc0ab (2020-08-01): Fix use of deprecated flask.ext imports

 flask.ext has been deprecated for years now and has been logging
 a warning to console ever since. OctoPrint 1.4.1 had to update
 the flask dependencies due to security considerations and with the
 new flask version the old compatibility layer still allowing
 flask.ext to work got removed.

 See also the discussion in OctoPrint/OctoPrint#3638

 Closes #269